lgsvl/simulator #1202: Camera position

andressa-suiti, 2021-01-06:
Hi. How can I get the camera coordinates in relation to the car's coordinate system? I'm using the Lincoln MKZ 2017 in my tests. Thanks :)

EricBoiseLGSVL, 2021-01-06:
@luqiang21 @hadiTab ?

luqiang21, 2021-01-06:
@andressa-suiti The vehicle's sensor config has `transform` for each sensor in world frame. You can use them to get relative relations between any two sensors. And depending on your own definition for car's coordinate system, you need to convert from unity coordinate to your car's coordinate as well.

EricBoiseLGSVL, 2021-01-06:
Thanks @luqiang21 

@andressa-suiti Unity world coordinate system is left-handed, y-up.  The sensors are positioned as child objects relative to the BaseLink object.  This object is at the origin of the vehicle mesh.  Unity needs meshes to have a zero'd origin.  This is at the bottom center of the mesh bounds.  If you have a vehicle you create, you can create and position the base link object where you need.  If not, it will be at this origin.  If you look at the docs, you can also position a sensor relative to another.

@luqiang21 is explaining that these Unity world space coordinates need to be converted to your coordinates.  Here is an example of ROS conversion [here](https://github.com/siemens/ros-sharp/wiki/Dev_ROSUnityCoordinateSystemConversion).

Here are some images showing the coordinates in Unity, this visualization is available in the UI panel in simulator.
[image]
[image]