lgsvl/simulator #1204: if there are two trrafic lights on one pole, then we need to create one or two?

Addi2020, 2021-01-06:


EricBoiseLGSVL, 2021-01-06:
@Addi2020 We currently support multiple traffic lights but they are the same state.  On the next release, we will provide many different traffic light models and multiple arrows/X/Uturn emission textures.
1) We do not yet have the logic for different states on the same roadway. You will need to create this logic or wait until we are able to implement
1.2) sounds like an issue with apollo.  We annotate the exact dimensions of the signal mesh. right @luqiang21 ?

luqiang21, 2021-01-06:
@EricBoiseLGSVL You are right. The overlapping issue should be an issue with apollo. @Addi2020 Did you try to annotate both traffic lights and change the `Signal Type` to `MIX_3_HORIZONTAL`?

closed 2021-07-29 by Addi2020