lgsvl/simulator #1206: Unable to load Maps, vehicles and simulation ( version 2020.06 )

prakyaths, 2021-01-07:
[image]
[image]
Following are the system specifications:
Memory: 11.7 GiB
Processor: Intel® Core™ i7-4770 CPU @ 3.40GHz × 8 
Graphics: GeForce GTX 1080/PCIe/SSE2
[image]

ntutangyun, 2021-01-07:
how about add the maps and vehicles using the add button

EricBoiseLGSVL, 2021-01-07:
@prakyaths Are you running the Simulator from the Unity Editor?  If so, it won't grab the default assets.
1. Navigate to the AppData folder for Simulator.  C:\Users\YourName\AppData\LocalLow\LG Silicon Valley Lab.  Linux is home/.config/unity3d/DefaultCompany/[project] 
2. Delete this folder
3. Download the simulator binary we provide [here](https://github.com/lgsvl/simulator/releases/tag/2020.06)
4. Run the binary and and open WebUI. Defaults assets will download.

One thing, with a 1080 GPU you will have performance issues running Apollo and Simulator on the same machine.

Another thing, like @ntutangyun commented, this isn't needed, you can clone the open source maps and vehicle, build the bundles and point the WebUI at them.  You can also make your own maps and vehicles.  I recommend you look at the repos provided first to see how they are constructed.