lgsvl/simulator #1207: Jaguar and San Francisco map missing meshes??

duongtran11, 2021-01-08:
I just downloaded the  vehicle and map assets from github and put it in the vehicle and environment folders as described in the instructions. But when I load them in unity the meshes seem to be missing
[image]
[image]

EricBoiseLGSVL, 2021-01-08:
It looks like gitlfs might have messed up.  Close unity and run git lfs pull.  Open unity.  If this doesn't work, delete the repo and re clone.

duongtran11, 2021-01-12:
Just figured out this only happens when I downloaded the .zip file. Everything is fine when I clone it using git clone. But before that I did reinstall gitlfs

closed 2021-01-12 by duongtran11

EricBoiseLGSVL, 2021-01-12:
Yes, the zip file doesn't have the assets, you must clone and have git lfs installed.