lgsvl/simulator #1209: /lgsvl/gnss_odom quaternion.w != 1 initially

henrychoi, 2021-01-08:
Hi, I am using the LGSVL simulator 2020.06 downloaded from the LGSVL website, along with the lgsvl_bridge installed on Ubuntu 20.04 LTS + ROS2 foxy with this command
```bash
sudo apt install ros-$ROS_DISTRO-lgsvl-msgs ros-$ROS_DISTRO-lgsvl-bridge
```
I am running Autoware.auto on the same computer.
With `ros2 topic echo /lgsvl/gnss_odom`, I see the following output:
```bash
henry@ade:~/AutowareAuto (henry/ms3-1 * u+3)$ ros2 topic echo /lgsvl/gnss_odom
1610121997.023352 [0]       ros2: using network interface enp3s0 (udp/192.168.2.5) selected arbitrarily from: enp3s0, docker0
header:
  stamp:
    sec: 1610122028
    nanosec: 767735040
  frame_id: odom
child_frame_id: base_link
pose:
  pose:
    position:
      x: -58.074466705322266
      y: -41.044307708740234
      z: -19.293531894683838
    orientation:
      x: -7.4505797087454084e-09
      y: 1.3341291404955768e-09
      z: 0.9930480122566223
      w: -0.11771061271429062
  covariance:
  - 0.0001
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0001
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0001
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0001
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0001
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0001
twist:
  twist:
    linear:
      x: 0.0
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: -0.0
  covariance:
  - 0.0001
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0001
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0001
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0001
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0001
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0001
```

Autoware uses the pose for localization, and I don't understand why the odometer sensor would be outputting some rotation value from the very beginning (wherever the vehicle is spawned).  My expectation is that it should be [0,0,0,1] at the beginning.  I wondered whether this postis a non-delta reporting, but the position doesn't make a lot of sense either.

EricBoiseLGSVL, 2021-01-08:
@henrychoi We have some fixes in the next release.  This is fixed or is a new issue @hadiTab ?

henrychoi, 2021-01-09:
Thanks @EricBoiseLGSVL .  How about I use the twist meanwhile?

hadiTab, 2021-01-12:
@henrychoi the sensor has a confusing name. It does *not* a pose estimate calculated from wheel odometers. It's a pose estimate that incorporates IMU data to end up providing a position in ENU coordinates (instead of lat/lon) and orientation in the form of a quaternion instead of a simple heading angle. The name comes from a message in Apollo by the same name.