lgsvl/simulator #1210: Controllable lights

fabvio, 2021-01-08:
Hi, thanks for your work.
I have a doubt regarding the controllability of street objects as light poles. Is it possible to turn the poles lights off at night? Is it possible for car headlights as well?
Thanks in advance.

EricBoiseLGSVL, 2021-01-09:
You could replace a light in the scene with a controllable or create a class and attach it to the object.  Really depends on how you plan on toggling this behavior.

fabvio, 2021-01-09:
So I guess it's going to require some modifications to the simulator. Why so? I've seen from promotional videos that all lights automatically toggle after a certain hour (6pm could it be?).

I would have to extract night images with every light in the scene alternatively turned off via the API. So there isn't a way to easily expose that?

EricBoiseLGSVL, 2021-01-11:
So you want custom Python API control to turn off lights on vehicles and street lights individually, correct?  This is not a simple thing to do and will require modification of the source code.
1) Create a Python API command in the simulator and python api to toggle vehicle lights.  You will need to target agents that you have added to the scene.
2) Modify NPC/EGO control classes to include an override method to turn off their lights when api calls are received.
3) Create simulator API command and Python API classes to search for TimeOfDay.cs light,  You can then iterate through the collection and enable/disable the light as needed.

If you are making your own environment, you can create a controllable object that has the API and simulator logic.  Controllables have policies that can be set via the python api and were designed for this exact purpose.  This issue is that exiting maps (esp one that are not open sourced) do not contain this custom logic.

If we have interest in this custom behavior, we can look into modifying the lights in a scene to make them controllable.  So far we have only seen requests for signal lights.

fabvio, 2021-01-11:
Yes I guess it's kind of a niche application :) thanks for the tips!

closed 2021-01-11 by fabvio