lgsvl/simulator #1211: Question about adding custom callbacks

Yimags, 2021-01-09:
Hello all,

I am now running lgsvl with Apollo and saving sensor data from cyberRT. How could I add callback so that it could rise when certain data were transferred? For example, when certain velocity or certain position of ego vehicle are reached. Thx in advance!

EricBoiseLGSVL, 2021-01-11:
@Yimags I would create a new sensor plugin or modify the sensor that is sending this data.  When the flag is set, send the custom callback.

Yimags, 2021-01-18:
@EricBoiseLGSVL Thx for the guidance! I think I may only need to make use of the existing sensors. And I found a example like following and found it really difficult to interpret. Could you maybe elaborate on how to make use of IMU data in this function? 
```
def onCustom(agent, kind, context):
  if kind == "comfort":
    print("Comfort sensor callback!", context)
  else:
    # ignore other custom callbacks
    pass

```

EricBoiseLGSVL, 2021-01-18:
@Yimags I would make the change in the sensor itself.  Here is an example in ComfortSensor.cs.  When a threshold is reached, the sensor adds a custom callback (event).  Then this event is sent via APIManager in fixed update.
```
if (ApiManager.Instance != null)
{
   ApiManager.Instance.AddCustom(transform.parent.gameObject, "comfort", jsonData);
}
```

Yimags, 2021-01-20:
@EricBoiseLGSVL Hello, I followed [your instruction of creating a new 3DGroundtruth sensor](https://github.com/lgsvl/simulator/issues/663#issuecomment-593730687) but stuck at step 6. In my case, I am trying to modify IMU sensor. How could I rename the class to avoid compile errors? Errors show as follows:
[image]

EricBoiseLGSVL, 2021-01-20:
Easiest way to dupe a sensor
In unity editor, select the sensor class
Hit Ctrl+D, this duplicates the class and names it ImuSensor1.cs.  This will have compile errors
Open the ImuSensor1.cs and rename the class name to ImuSensor1 (Inside the class)
Save and recompile.
All errors will be gone.  I suggest you change the class name to something better than ImuSensor1 but this is the way.

To make this new class an actual sensor for a simulation...
Create a new gameobject in any scene, add the sensor class to the object (zero the transform please :) ) and name the gameobject.
Drag the gameobject into Assets->Prefabs->Sensors in the Project Panel.  This creates a prefab. delete the prefab in the scene

Then drag the new sensor into the Resources->RuntimeSettings SensorPrefabs array that you increase index by one.

Then reload or open close unity and it will load this new sensor.  You can run loader scene and include a sim with the new sensor json and it will load it so you can debug further.

Yimags, 2021-01-20:
I followed the instructions you provided to the final step and successfully created the new prefab in Assets->Prefabs->Sensors and named it as 'imu_callback'. When I try to use it in json configuration of vehicle in WedUI, I put it as follows.
```
  {
    "type": "imu_callback",
    "name": "imu_callback",
    "params": {
      "Topic": "/apollo/sensor/gnss/imu",
      "Frame": "imu",
      "CorrectedTopic": "/apollo/sensor/gnss/corrected_imu",
      "CorrectedFrame": "imu"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": -1.348649,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  
```
I got an error of 'Invalid: Unknown sensor type imu_callback for vehicle'. How could I fix it?

EricBoiseLGSVL, 2021-01-21:
Did you change this also?
```
[SensorType("IMU", new[] { typeof(ImuData) })] to [SensorType("imu_callback", new[] { typeof(ImuData) })]
```

FYI, C# Class names should be PascalCase.  You may run into issues with _'s

Yimags, 2021-01-21:
Yes, and according to your suggestion, I further change it to `[SensorType("ImuCallback", new[] { typeof(ImuData) })]`. However, the error remains. Does it necessary to build it with map and vehicle to make it functional?

EricBoiseLGSVL, 2021-01-21:
@Yimags yes if you add the sensor to the default ones, you need to make a new binary.

Yimags, 2021-01-21:
@EricBoiseLGSVL Sorry that I am actually new to Unity and lgsvl. Could you explain a bit more on what do you mean by making a new binary and how to make it?

EricBoiseLGSVL, 2021-01-21:
@Yimags no worries.  In editor go to Simulator -> Build... At the bottom you can build the simulator.  Be sure your editor has Windows and Linux build modules.

Yimags, 2021-01-21:
@EricBoiseLGSVL  Thx for the explanation. I tried also to make a new sensor plugin by following the ducumentation and buit the changes to simulator. The result remains the same: `Unknown sensor type`. Even when I changed the `[SensorType("IMU", new[] { typeof(ImuData) })]` in original ImuSensor.cs file to  `[SensorType("IMU New", new[] { typeof(ImuData) })]`, in json I could still only use type:'IMU'. So I guess I missed some steps to apply the changes I made in .cs scripts?

EricBoiseLGSVL, 2021-01-21:
Wait, did you make a plugin in Assets->External->Sensors->ImuNew? or in the source code in Assets->Scripts->Sensors->ImuNew?
If the former you need to make a new sensor plugin with build and then point WebUI at the new plugin asset bundle
If the latter you need to make a new simulator binary and the sensor becomes a default sensor.  Which one did you try?

Yimags, 2021-01-21:
Actually I tried them both. Now I found the reason. I was keeping using the downloaded `lgsvlsimulator-linux64-2020.06`  instead of the self-built simulator. I turn to the built version then both of these two methods work well.

However, it seems to me that the self-built simulator is not compatible with some assets. It keeps showing error 'AssetBundle out of date'  for the same simulation where the downloaded version works well. How could I fix this?

EricBoiseLGSVL, 2021-01-21:
I suggest you use the 06 binary and use the sensor plugin version.  This will keep compatibility with other assets.

Yimags, 2021-01-21:
Thanks for the suggestion! But how could I build sensor plugin alone without the simulator? Once I tickle 'Build Simulator' and specify the path, 06 version simulator will be covered with the self-built one.

EricBoiseLGSVL, 2021-01-22:
Since you placed the sensor code in the folder Assets/External/Sensors/ImuNew you can open the menu, Simulator->Build...
Check the sensor and build (no need to build a new sim binary)  Be sure all code and assets are in the folder, not in simulator code base.  Then look in the repo for the AssetBundles/Sensors/ for your bundle, make sure it isn't 0k.  Now when you run Editor it will load the new sensor on startup.  If you want to use the 06 binary with the new sensor, copy and place in the AssetBundles/Sensors folder like the repo.  Docs [here](https://www.lgsvlsimulator.com/docs/sensor-plugins/)

Yimags, 2021-01-22:
Everytime when I move the built sensor plugin to AssetBundles/Sensors/ in 06 binary, it shows a problem of 'Invalid: Object reference not set to an instance of an object' for every sim. It  recovers only when I remove the new plugin.  Did I miss some steps?

[image]

EricBoiseLGSVL, 2021-01-22:
No it should work.  Can you post the sensor code here and I'll and @mark-gerow-lge for more help

Yimags, 2021-01-22:
Yeah sure. Here it is.(https://github.com/lgsvl/simulator/files/5858489/sensor.bundle.zip)

EricBoiseLGSVL, 2021-01-22:
Here is your issue.
```
[SensorType("IMU Plugin", new Type[] { typeof(ImuData) })]
internal class ImuCustom : SensorBase
{
}
```
should be
```
[SensorType("ImuCustom", new Type[] { typeof(ImuData) })]
internal class ImuCustom : SensorBase
{
}
```

then the sensor JSON should be
```
{
    "type": "ImuCustom",
    "name": "User defined doesn't matter",
    "params": {
      "Topic": "/imu",
      "Frame": "imu"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
}
```
the type refers to SensorType.  Was this different in your JSON?

Yimags, 2021-01-22:
Actually I use "IMU Plugin" also in JSON, and it works well in self-built binary. But it shows error when I copied it to 06 binary.

EricBoiseLGSVL, 2021-01-22:
Hmm, this might be a bug then.  I know we have a bunch of fixes in the next release.  I'll see if I can track it down.

EricBoiseLGSVL, 2021-01-22:
Please try to use "ImuCustom" as the sensorType and in the JSON and post if it works, thanks

Yimags, 2021-01-22:
I change the sensorType and it is all good in self-built binary.

But it is kinda weird. Even though I didn't define the new plugin in JSON, the error will still occur as soon as I copy the plugin bundle in 06 binary. Did it work well in your 06 binary?

EricBoiseLGSVL, 2021-01-22:
Can you please zip the sensor code so I can try to build locally.  Maybe the bundle building is having issues

EricBoiseLGSVL, 2021-01-22:
I guess the only way forward is to make bundles for the other assets and point webui at all the new assets from the new binary.  You can still post the sensor code and I'll try to rebuild also

Yimags, 2021-01-22:
Here. Thx in advance! [ImuCustom.zip](https://github.com/lgsvl/simulator/files/5858930/ImuCustom.zip)

EricBoiseLGSVL, 2021-01-22:
no problem.  Did you make a prefab for the sensor?

Yimags, 2021-01-22:
Err yes, but the simplest prefab. You could find it here. [ImuCustom.zip](https://github.com/lgsvl/simulator/files/5858970/ImuCustom.zip)

I save the sensor script and no error occurs so I guess it is compiled?
I add only the callback parts similar to Comfort sensor. For IWriter and IBridge I think they exist in original ImuSensor.cs scripts and I just copied them.

EricBoiseLGSVL, 2021-01-22:
No, this had multiple errors and would not compile.  What source code are you using?  This has to be built with 06 release code.  are you using a different version?  Please be sure you have 06 release code cloned

Here is the fixed and working sensor source and bundle to test.  Import custom asset package into unity and then copy out the included bundle and move to the 06 binary AssetBundles folder to test.

[ImuCustomFixed.zip](https://github.com/lgsvl/simulator/files/5859067/ImuCustomFixed.zip)

Yimags, 2021-01-22:
Ahhh, now I notice you didn't merge the 06 source with master branch. I will clone 06 release, try the new assets and  post the result later.

Yimags, 2021-01-25:
@EricBoiseLGSVL  Hello, after having correct 06 version cloned. I tried your code and it works perfectly. And I also found that, unlike you said, `sensortype `and `classname ` in sensor script are not necessarily to the the same. Just to clarify. 

In addition, I have a question about how I could add callbacks to channels which is not published by lgsvl? For example the channel published by apollo( e.g., control, routing). Shall I similarly create a sensor plugin for that, I guess?

EricBoiseLGSVL, 2021-01-25:
@Yimags excellent.  Glad it is working.  Yes, you can subscribe to a topic and have logic in the sensor

Yimags, 2021-01-25:
There is no related sensor script for my desired channel (Routing for example), so how could I create a new sensor type? Is it same as the protobuf format in Apollo ,and I need to transform it to data.cs script?

I also find cyber bridge plugin which could be used to save data from bridge. How it is different from sensor plugin and should I use it instead of sensor plugin?

EricBoiseLGSVL, 2021-01-25:
I'll ping @hadiTab on this.  @Yimags please sum up exactly what you are trying to do so Hadi can point you in the right direction

Yimags, 2021-01-26:
Thx! Actually, I want to accomplish 2 tasks.
1. Add callback to apollo routing function. Once routing in apollo is finished, callback will be invoked.
2. Create a [RSS](https://intel.github.io/ad-rss-lib/) sensor like what [carla](https://www.youtube.com/watch?v=UxKPXPT2T8Q) do. Not so complex visual effects, only need to show a boolean result in sensor.

closed 2021-04-14 by Yimags

EricBoiseLGSVL, 2021-04-14:
1) Not sure on this, sounds like you want to modify Apollo/Dreamview right?  On simulator side, you can create a sensor plugin that receives the callback from dreamview and then send a custom api callback and/or topic publish that it is received and then run your logic. 
2) This is pretty simple.  Look to our docs on creating a sensor plugin.  Then look to Unity apis to raycast or measure distance to NPCs and output result to topic and/or Vizualize method. Search "Unity get distance to gameobject"