lgsvl/simulator #1212: Pointcloud map does not load in simulation when loaded with python-api

gjonik-fhac, 2021-01-11:
Hello,

ive been trying to automate the generation of training data for a 3d detection algorithm and would like to use a .las laserscan of a certain place in my area as map. The import with the pointcloud importer works fine. When I start an interactive simulation with no api i can see the map and my agents collide as expected. But when I load it through the python-api the whole scene is empty.

EricBoiseLGSVL, 2021-01-11:
@gjonik-fhac This is a bug and has been fixed in the next release.

gjonik-fhac, 2021-01-11:
@EricBoiseLGSVL Is it going to be released in the near future? Can I find a release schedule or plan somewhere?

EricBoiseLGSVL, 2021-01-12:
I'll see if I can find the fix and post here if it can be.

gjonik-fhac, 2021-01-12:
That would be very nice of you!

EricBoiseLGSVL, 2021-01-26:
Sorry, the fix requires multiple changes.  We will have to wait until next release.