lgsvl/simulator #1213: About the vector map

qiongwu1003, 2021-01-12:
Hi,
LGSVL team

I run the Autoware.AI with LGSVL simulator, refer this [instruction](https://www.lgsvlsimulator.com/docs/autoware-instructions/).
And now I want to use my own map, so I need to draw my own vector map.
I found a third-party tool "[MapToolBox](https://github.com/autocore-ai/MapToolbox)" that can be used to draw the vector map, but it only supports op_global_planner mode. Under above [instruction](https://www.lgsvlsimulator.com/docs/autoware-instructions/) the vector map has problems.
I also tried [Vector Map Build](https://tools.tier4.jp/), it exports txt files but not csv files. Otherwise it is difficult to use.

So may I know what tools you use to draw vector maps?
Or if LGSVL simulator can run with Autoware.AI under the op_global_planner mode? I am new with Autoware.Ai, and I do not found any demo.

Thanks. Regards

luqiang21, 2021-01-12:
@qiongwu1003 Our map exporter can export Autoware Vector Map. Did you try it?

qiongwu1003, 2021-01-13:
@luqiang21 

The "map exporter" do you mean the below :
[image]

I am using release 202006 simualtor, and I can not find export Autoware Vector Map tab.

qiongwu1003, 2021-01-13:
@luqiang21 

Sorry, I found the export Autoware Vector Map tab.
Thank you very much.

closed 2021-01-13 by qiongwu1003

dean1314, 2021-07-25:
have you sovled this issue yet ,  I'm also met similar issue.  I'm trying to use op_global_planner to generate global paths on BorregasAve map provided by LGSVL/Autoware-data,  but it took error in Extract vector map process ...