lgsvl/simulator #1215: apollo与LGSVL联合模拟

hxj0316, 2021-01-14:
你好，我想请问一下大家都是使用哪个apollo版本和LGSVL进行联合模拟的呀？5.5和6.0版本是否支持与LGSVL的联合模拟呢？
还有，大家都是用的什么配置来运行的LGSVL？我看有人说2080的显卡好像会有点问题是吗。

luqiang21, 2021-01-15:
> 你好，我想请问一下大家都是使用哪个apollo版本和LGSVL进行联合模拟的呀？5.5和6.0版本是否支持与LGSVL的联合模拟呢？
还有，大家都是用的什么配置来运行的LGSVL？我看有人说2080的显卡好像会有点问题是吗。

> Hello, I would like to ask which apollo version and LGSVL do you use for simulation? Do versions 5.5 and 6.0 support simulation with LGSVL?
Also, what configuration do you use to run LGSVL? I think some people say that the 2080 graphics card seems to be a problem, right?

你好，Apollo 的各个版本都是支持的，只不过5.0（包括5.0）之前的版本，lgsvl有Apollo的fork （比如 [apollo-5.0](https://github.com/lgsvl/apollo-5.0)），你需要用fork来跑。5.0之后的版本，因为cyber bridge已经merged到Apollo的master了，所以进行联合仿真直接用Apollo官方repo就行。文档里面有针对不同版本的教程: [latest master](https://www.lgsvlsimulator.com/docs/apollo-master-instructions/), [apollo-5.0](https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/), [apollo-3.0](https://www.lgsvlsimulator.com/docs/apollo-instructions/)。要注意的是目前apollo master perception可能有问题，你可以用[modular testing sensors](https://www.lgsvlsimulator.com/docs/modular-testing/)来避免运行perception。
配置的话，几乎所有显卡都可以跑LGSVL，只是Apollo对显卡有限制。

Hello, LGSVL supports all apollo versions. Before apollo 5.0 (including 5.0), you need to use our fork (e.g. [apollo-5.0](https://github.com/lgsvl/apollo-5.0)) to run with the simulator. After 5.0, you need to use apollo master to run with the simulator since cyber_bridge is merged to Apollo. The documentation has tutorials for different apollo versions: [latest master](https://www.lgsvlsimulator.com/docs/apollo-master-instructions/), [apollo-5.0](https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/), [apollo-3.0](https://www.lgsvlsimulator.com/docs/apollo-instructions/). Note: current apollo master perception may have problems, you can use [modular testing sensors](https://www.lgsvlsimulator.com/docs/modular-testing/) to avoid running it.
For the GPU configuration to run LGSVL, there is no limit. There are constraints to run apollo which you need to check.

bluelml, 2021-01-21:
Hi @luqiang21 , I tried the following releases of apollo( latest master, lgsvl apollo-5.0, pr6), but none can work. 
The main problems were traffic light issue or send routing request not working. 

Could you tell me a certain version which can work and what necessary  should I modify for this version?

Thank a lot.

luqiang21, 2021-01-21:
@bluelml If your perception or traffic light doesn't work with lgsvl apollo-5.0, it could be your GPU memory is not enough to run both simulator and apollo. 
For "send routing request not working", did you see a red route in dreamview after you set destination?

bluelml, 2021-01-23:
@luqiang21 
1. I am using RTX8000 with 48GB memory. So this should't be the root cause. 
2. If I using lgsvl apollo-5.0, the perception model did not work. 
2. Yes, I can see a red route, but the cat stay here do nothing  If I using apollo pr6.
So I was confused with version of apollo.

luqiang21, 2021-01-24:
@bluelml Apollo-5.0 all modules are working with LGSVL simulator for me. You should check apollo logs to see which modules are not working properly. If you can see a red route, you should check if `perception`, `prediction` or `planning`'s logs have some useful clue.