lgsvl/simulator #1218: Apollo can not launch perception and GPS

DavidSu0902, 2021-01-18:
system: Ubuntu18.04
memory: 32G
processor: i9-9900k
graphics: GeForce RTX 2080super
apollo brach: apollo-5.0 folked by lgsvl
senor configuration: apollo-5.0 configuration provided by the document

I followed the installation guide and meets some problem.
I can see the car in deamview and I can control it with keyboard. But I can not launch perception module and GPU module.

 
[image]

DavidSu0902, 2021-01-18:
I used ./scripts/gps.sh start_fe to see what happened.
[image]
I meet:
[gnss   ]  E0118 16:42:08.208170  8151 serial_stream.cc:127] [mainboard] Open device /dev/novatel0 failed, error: No such file or directory
[gnss   ]  E0118 16:42:08.208199  8151 raw_stream.cc:332] [mainboard] data stream connect failed.
[gnss   ]  E0118 16:42:08.208207  8151 raw_stream.cc:273] [mainboard] gnss driver connect failed.
[gnss   ]  E0118 16:42:08.208241  8151 gnss_component.cc:42] [gnss] Init gnss stream failed
[gnss   ]  E0118 16:42:08.208250  8151 component.h:155] [mainboard] Component Init() failed.
[gnss   ]  E0118 16:42:08.208272  8151 serial_stream.cc:127] [mainboard] Open device /dev/novatel0 failed, error: No such file or directory
[gnss   ]  E0118 16:42:08.208281  8151 serial_stream.cc:127] [mainboard] Open device /dev/novatel0 failed, error: No such file or directory
[gnss   ]  E0118 16:42:08.209471  8151 module_controller.cc:59] [mainboard] Failed to load module: /apollo/modules/drivers/gnss/dag/gnss.dag
[gnss   ]  E0118 16:42:08.209856  8151 mainboard.cc:43] [mainboard] module start error.

DavidSu0902, 2021-01-18:
I used ./scripts/perception.sh start_fe
[image]
[mainboard_default_8509]  E0118 16:44:58.467396  8559 static_transform_component.cc:59] [static_transform] Extrinsic yaml file is noe exists: /apollo/modules/drivers/velodyne/params/lidar16_front_up_novatel_extrinsics.yaml
[mainboard_default_8509]  E0118 16:44:58.467692  8559 static_transform_component.cc:59] [static_transform] Extrinsic yaml file is noe exists: /apollo/modules/drivers/velodyne/params/lidar16_front_novatel_extrinsics.yaml
[mainboard_default_8509]  E0118 16:44:58.467880  8559 static_transform_component.cc:59] [static_transform] Extrinsic yaml file is noe exists: /apollo/modules/drivers/velodyne/params/lidar16_rear_left_novatel_extrinsics.yaml
[mainboard_default_8509]  E0118 16:44:58.468046  8559 static_transform_component.cc:59] [static_transform] Extrinsic yaml file is noe exists: /apollo/modules/drivers/velodyne/params/lidar16_rear_right_novatel_extrinsics.yaml
[mainboard_default_8509]  E0118 16:44:58.471202  8559 static_transform_component.cc:59] [static_transform] Extrinsic yaml file is noe exists: /apollo/modules/perception/data/params/radar_rear_extrinsics.yaml
[mainboard_default_8509]  E0118 16:44:58.680459  8737 node_manager.cc:101] [mainboard] this process will be terminated due to duplicated node[static_transform], please ensure that each node has a unique name.
[mainboard_default_8509]  
[motion_service]  [NVBLAS] NVBLAS_CONFIG_FILE environment variable is NOT set : relying on default config filename 'nvblas.conf'
[motion_service]  [NVBLAS] Cannot open default config file 'nvblas.conf'
[motion_service]  [NVBLAS] Config parsed
[motion_service]  [NVBLAS] CPU Blas library need to be provided
[cyber_launch_8509] ERROR Process [mainboard_default_8509] has finished. [pid 8559, cmd mainboard -d /apollo/modules/transform/dag/static_transform.dag -p mainboard_default_8509 -s CYBER_DEFAULT].
[cyber_launch_8509] INFO All processes has died.
[cyber_launch_8509] INFO Cyber exit.
[cyber_launch_8509] INFO All processes have been stopped.
[mainboard_default_8510]  E0118 16:44:59.628058  8736 node_manager.cc:101] [mainboard] this process will be terminated due to duplicated node[camera_front_6mm_decompress], please ensure that each node has a unique name.
[mainboard_default_8510]  E0118 16:44:59.628192  8736 node_manager.cc:101] [mainboard] this process will be terminated due to duplicated node[camera_front_12mm_decompress], please ensure that each node has a unique name.
[mainboard_default_8510]  E0118 16:44:59.628230  8736 node_manager.cc:101] [mainboard] this process will be terminated due to duplicated node[camera_left_fisheye_decompress], please ensure that each node has a unique name.
[mainboard_default_8510]  E0118 16:44:59.628262  8736 node_manager.cc:101] [mainboard] this process will be terminated due to duplicated node[camera_right_fisheye_decompress], please ensure that each node has a unique name.
[mainboard_default_8510]  E0118 16:44:59.628293  8736 node_manager.cc:101] [mainboard] this process will be terminated due to duplicated node[camera_rear_6mm_decompress], please ensure that each node has a unique name.
[mainboard_default_8510]  
[cyber_launch_8510] ERROR Process [mainboard_default_8510] has finished. [pid 8557, cmd mainboard -d /apollo/modules/drivers/tools/image_decompress/dag/image_decompress.dag -p mainboard_default_8510 -s CYBER_DEFAULT].

DavidSu0902, 2021-01-18:
If I shut down dreamview, I can see gnss channel in cyber_monitor
[image]

DavidSu0902, 2021-01-18:
I also meet this using  cyber_launch start modules/perception/production/launch/perception_camera.launch
[image]

ntutangyun, 2021-01-18:
dreamview 里面 mode 选 Mkz Lgsvl 试试？

ntutangyun, 2021-01-18:
apollo-5.0 forked version is no longer being maintained. please use official apollo 6.0 or pre6 instead.

FitashUlHaq, 2021-01-18:
Apollo 5.0 does not work with GeForce RTX 2080

DavidSu0902, 2021-01-18:
@ntutangyun I swtich to apollo latest version, now I can not launch any module...
[image]

ntutangyun, 2021-01-18:
what error are u facing when launching the modules from terminal, say transform module

DavidSu0902, 2021-01-19:
@ntutangyun 
[image]
localization]  F0119 09:56:02.920255  7575 localization_lidar.cc:62] Reflectance map folder is invalid!
[localization]  *** Check failure stack trace: ***
[cyber_launch_7568] ERROR Process [localization] has died [pid 7575, exit code -6, cmd mainboard -d /apollo/modules/localization/dag/dag_streaming_rtk_localization.dag -p localization -s CYBER_DEFAULT].
[cyber_launch_7568] INFO All processes has died.
[cyber_launch_7568] INFO Cyber exit.

ntutangyun, 2021-01-19:
> @ntutangyun
>
> [image]
>
> localization]  F0119 09:56:02.920255  7575 localization_lidar.cc:62] Reflectance map folder is invalid!
>
> [localization]  *** Check failure stack trace: *** [...]

you are using msf_localization. 

to work with similator, you need to use rtk_localization instead.

DavidSu0902, 2021-01-20:
> you are using msf_localization.
>
> to work with similator, you need to use rtk_localization instead.

How can I swtich to rtk_localization?

DavidSu0902, 2021-01-20:
> rtk_localization

and in my log, the error info says dag_streaming_rtk_localization...