lgsvl/simulator #1219: Radar issue in release-2020.06

endland, 2021-01-18:
Hello. While I'm trying to use the radar sensor using ROS2 bridge in latest binary version of lgsvl simulator 2020.06, I found out it doesn't support the radar sensor. So I started to build from source and add just one line of code in simulator/Assets/Scripts/Bridge/Ros2/Ros2BridgeFactory.cs

            RegPublisher<Detected3DObjectData, Lgsvl.Detection3DArray>(plugin, Ros2Conversions.ConvertFrom);
            RegPublisher<Detected2DObjectData, Lgsvl.Detection2DArray>(plugin, Ros2Conversions.ConvertFrom);
 --->     RegPublisher<DetectedRadarObjectData, Lgsvl.DetectedRadarObjectArray>(plugin, Ros2Conversions.ConvertFrom);
            RegPublisher<SignalDataArray, Lgsvl.SignalArray>(plugin, Ros2Conversions.ConvertFrom);

After that, I can see the radar sensor in the simulator and the topic is also published via lgsvl_bridge. 
But there is no message in radar topic and "NullReferenceException: Object reference not set to an instance of an object" error is pouring in INFO tap of simulator. 
[image]

Could you give any advice to use the radar sensor correctly for ROS2 in lgsvl simulator?

EricBoiseLGSVL, 2021-01-18:
@endland Did you add conversion function in Ros2Conversions.cs?  Can you run in editor so we can see the call stack? Adding @hadiTab

endland, 2021-01-19:
@EricBoiseLGSVL  Isn't it already conversion function in Ros2Conversions.cs like the below?  Should I add another code for it? 
How can I run it in editor? Please, show me a guide for it. Many thanks. 

 public static Lgsvl.DetectedRadarObjectArray ConvertFrom(DetectedRadarObjectData data)
        {
            var r = new Lgsvl.DetectedRadarObjectArray()
            {
                header = new Ros.Header()
                {
                    stamp = Convert(data.Time),
                    frame_id = data.Frame,
                },
            };

            foreach (var obj in data.Data)
            {
                r.objects.Add(new Lgsvl.DetectedRadarObject()
                {
                    sensor_aim = ConvertToVector(obj.SensorAim),
                    sensor_right = ConvertToVector(obj.SensorRight),
                    sensor_position = ConvertToPoint(obj.SensorPosition),
                    sensor_velocity = ConvertToVector(obj.SensorVelocity),
                    sensor_angle = obj.SensorAngle,
                    object_position = ConvertToPoint(obj.Position),
                    object_velocity = ConvertToVector(obj.Velocity),
                    object_relative_position = ConvertToPoint(obj.RelativePosition),
                    object_relative_velocity = ConvertToVector(obj.RelativeVelocity),
                    object_collider_size = ConvertToVector(obj.ColliderSize),
                    object_state = (byte)obj.State,
                    new_detection = obj.NewDetection,
                });
            }

            return r;
        }

endland, 2021-01-19:
The problem seems like the process way of the detected objects. When no objects are detected, the radar topic is issuing like the below.  

$ ros2 topic echo /radar 
objects: []
---
header:
  stamp:
    sec: 1611036264
    nanosec: 727538432
  frame_id: ''
objects: []
---
header:
  stamp:
    sec: 1611036264
    nanosec: 807538176
  frame_id: ''
objects: []
---

But topic pauses when some objects are detected and simulator spits out "NullRefercenceException: Object reference not set to and instance of an object". 

Could you give me some advice to resolve this problem?  Thanks.

EricBoiseLGSVL, 2021-01-19:
@endland Does this behavior happen without your code changes?

endland, 2021-01-19:
Yes. I clone the release 2020.06 and add just one line of code in simulator/Assets/Scripts/Bridge/Ros2/Ros2BridgeFactory.cs

RegPublisher<DetectedRadarObjectData, Lgsvl.DetectedRadarObjectArray>(plugin, Ros2Conversions.ConvertFrom);

EricBoiseLGSVL, 2021-01-20:
no does the null error occur without your change to the code?

duanchengwen, 2021-04-08:
@endland i meet the same error, did you solve the problem?

duanchengwen, 2021-04-08:
> @endland i meet the same error, did you solve the problem?

i solved, 
[image]

EricBoiseLGSVL, 2021-04-08:
linking to #1349 and we are looking into this issue

endland, 2021-04-11:
@duanchengwen  Thanks for sharing your solution. I did it for ROS2. 

`        public static Lgsvl.DetectedRadarObjectArray ConvertFrom(DetectedRadarObjectData data)
        {
            var r = new Lgsvl.DetectedRadarObjectArray()
            {
                header = new Ros.Header()
                {
                    stamp = Convert(data.Time),
                    frame_id = data.Frame,
                },
                objects = new List<Lgsvl.DetectedRadarObject>(),
            };

            int count = 0;
            foreach (var obj in data.Data)
            {
                var det = new Lgsvl.DetectedRadarObject()
                
                {
                    id = count++,
                    sensor_aim = ConvertToVector(obj.SensorAim),
                    sensor_right = ConvertToVector(obj.SensorRight),
                    sensor_position = ConvertToPoint(obj.SensorPosition),
                    sensor_velocity = ConvertToVector(obj.SensorVelocity),
                    sensor_angle = obj.SensorAngle,
                    object_position = ConvertToPoint(obj.Position),
                    object_velocity = ConvertToVector(obj.Velocity),
                    object_relative_position = ConvertToPoint(obj.RelativePosition),
                    object_relative_velocity = ConvertToVector(obj.RelativeVelocity),
                    object_collider_size = ConvertToVector(obj.ColliderSize),
                    object_state = (byte)obj.State,
                    new_detection = obj.NewDetection,
                };
                
                r.objects.Add(det);
            }

            return r;
        }`

closed 2021-04-11 by endland