lgsvl/simulator #1220: Ground truth visualizer renders ground truth data in wrong plane

ar4development, 2021-01-18:
1. Set up ground truth sensor 
2. Set up ground truth visualizer sensor so that it subscribes the topic ground truth publishes to
3. Run simulation

EXPECTED: Rendered boxes fit the vehicles
OBSERVED: They are rendered in different plane. Like shown on screenshot.

[image]

**Sensors:**

```
  {
    "type": "3D Ground Truth",
    "name": "3D Ground Truth",
    "params": {
      "Frequency": 10,
      "Topic": "/test/ground_truth/detections_3d"
    },
    "transform": {
      "x": 0,
      "y": 1.975314,
      "z": -0.3679201,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "3D Ground Truth Visualizer",
    "name": "3D Ground Truth Visualizer",
    "params": {
      "Topic": "/test/ground_truth/detections_3d"
    },
    "transform": {
      "x": 0,
      "y": 1.975314,
      "z": -0.3679201,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  }
```

**Map:** https://assets.lgsvlsimulator.com/416285c369f9cd9862e2797818f39e9afe9c8f07/environment_CubeTown

**LGSVL version**: linux64-2020.06

EricBoiseLGSVL, 2021-01-19:
Maps were rotated to fix long standing issues with autoware.  Please see #1187 and #961.  Do you have the updated map?

ar4development, 2021-01-19:
@EricBoiseLGSVL The map I have mentioned in description seems to be the actual map from https://content.lgsvlsimulator.com/maps/cubetown/ (see `Asset Bundle` link)

EricBoiseLGSVL, 2021-01-19:
@ar4development Hmm, that is the right map.  Are you using updated HDMap being used?  @hadiTab doesn't users need to update map and hdmap.  Visualizer is returning what the AD stack detects.

ar4development, 2021-01-20:
@EricBoiseLGSVL Hi Eric,

I'm not very familiar with LGSVL. I've googled around and as far as I now understand (probably I'm wrong though) hd maps are somehow related to autoware stack. However I'm just looping Ground Truth data from simulator to the same simulator without any intermediate processing. Just reading from the topic that simulator writes to..

I believe that map bundle has to be enough for having everything rendered in proper coordinates.

EricBoiseLGSVL, 2021-01-20:
@ar4development Ok I see.  yes 3DGTVisualizer is not for simulator GT but the AD stack ground truth.  It is for Apollo and is expecting data to be rotated.

ar4development, 2021-01-20:
@EricBoiseLGSVL Alright. My goal is to evaluate boxes for the objects using some arbitrary algorithm. It takes the lidar points from LGSVL. I use GPS sensor to obtain position and orientation for EGO. Then I use that data to translate positions of detected objects from lidar space to map space.

I also want to render detected boxes in simulator.

Taking into account that the evaluations are performed in the space of the map which cannot the be correctly visualized how do I transform my objects to have them properly oriented? Or probably it is easier to apply transformation to the 3DGTVisualizer sensor? Which exact transformation should I perform?

EricBoiseLGSVL, 2021-01-20:
@hadiTab What would be the best solution?  Create a new Visualizer sensor that orients the data.  Is GroundTruth3D causing the rotation or is it visualizer?

zelenkovsky, 2021-02-12:
The issue is caused by coordinate system differences. These sensors expect different coordinate systems.
https://www.lgsvlsimulator.com/docs/simulator-coordinate-system/

I believe it could be fixed by adding an ENUM parameter to the Ground Truth Visualize Sensor, allowing to specify the coordinate system.