lgsvl/simulator #1222: HDRP version issue. There is no SetupGlobalParams in new HDRP version

xogur6889, 2021-01-19:
I'm trying to run this project with a new HDRP version. But in the new version, the customRender's hd.SetupGlobalParams (cmd, 0) function does not exist. I think it has been changed to UpdateShaderVariablesGlobalCB or something, can you help me apply it?

EricBoiseLGSVL, 2021-01-19:
@xogur6889 We do not support updating HDRP with latest simulator.  We separate HDRP and make edits to the source code to provide features.  That is why we include it in the Packages folder.  If you update HDRP, you run the risk of breaking other simulator functionality.  Why are you updating HDRP?  Is there a feature that you need or bugs that are causing an issue?

xogur6889, 2021-01-20:
> @xogur6889 We do not support updating HDRP with latest simulator. We separate HDRP and make edits to the source code to provide features. That is why we include it in the Packages folder. If you [...]

Hello!! Thank you for your quick reply. I'm Tae-hyeok, a Korean student.
I use the lidar sensor prefab and I want to apply shader to the camera and get information about position only. The reason for updating the HDRP version is that we use the Crest HDRP Asset to simulate with the lidar on a ship in a sea environment.
[image]

EricBoiseLGSVL, 2021-01-20:
@xogur6889 Very interesting project and this asset looks very good.  I'll see what we need to do to support 2019.4.18

xogur6889, 2021-01-21:
> @xogur6889 Very interesting project and this asset looks very good. I'll see what we need to do to support 2019.4.18

Thank you for your interest. **I think the HDRP version is important. I used the 10.2.2 version.**
The main points that have been modified so far are as shown in the picture below.
[image]
First, I did not use the SetupGlobalParams function (I tried to use it, but there was no difference). And I used the CopyFrom function to copy the main camera (I saw and followed the creation of the Preview Camera in the Editor) so that the camera appeared in the Game View and the position value came out right. However, this method had the problem of not working properly when other cameras were being rendered.

EricBoiseLGSVL, 2021-01-21:
If you look at the asset description in the Unity asset store, Crest says that it doesn't support multiple cameras.  You might need to contact the developers to see how to support this.

xogur6889, 2021-01-25:
> If you look at the asset description in the Unity asset store, Crest says that it doesn't support multiple cameras. You might need to contact the developers to see how to support this.

Hello! I'm Tae-Hyeok. Thank you for your kind answer.
Crest Asset uses one camera to render the sea as shown below. This seems to mean that they are not supporting multiple cameras.
[image]

However, there was no problem using multiple cameras with Crest Asset in the project itself. The picture below shows the depth map with a game view using a custom render.
[image]

The point is, I tested the Lidar Sensor in a very simple environment, but the problems that I mentioned continued to arise. The picture below is the environment that I tested.(Currently, only one camera is rendering to Game View and captures the status of proper operation.)
[image]

EricBoiseLGSVL, 2021-01-25:
@xogur6889 Again, we alter the HDRP package.  If you overwrite with a new version, it removes our code.  You can diff the folder in Packages to see the difference.  Lidar/Segmentation will not work unless you either add this code to the source or wait until we release an updated Unity project.

LiangHuangBC, 2022-11-23:
Any progress on this, and is there any doc I can know how to simulate a boat on the ocean?