lgsvl/simulator #1225: what is the xyz order for sensor pararmter transformation

qwetqwe, 2021-01-25:
In  document , I see Sensor JSON Options:
x is the position of the sensor along the x-axis
y is the position of the sensor along the y-axis
z is the position of the sensor along the z-axis
pitch is the rotation around the x-axis
yaw is the rotation around the y-axis
roll is the rotation around the z-axis
So what is the xyz order for pitch yaw roll

EricBoiseLGSVL, 2021-01-25:
Python uses Vector for rotation.  In Unity you can see the source code that uses Unity api Quaternion.Euler(Vector) to translate the Euler value to a Quaternion.  Is this what you are asking?  See Unity docs [here](https://docs.unity3d.com/ScriptReference/Quaternion.Euler.html).

qwetqwe, 2021-01-26:
Not what I mean. For Euler angle rotation, first rotate x-axis angle , next y-axis angle  ,then z-axis angle,and first z-axis angle, next x-axis angle,then y-axis angle will be different  posture.  So I want to know what is the xyz order

qwetqwe, 2021-01-26:
From your docs, I see
Euler angles are represented by three angle values for X, Y and Z that are applied sequentially. To apply an Euler rotation to a particular GameObject, each rotation value is applied in turn, as a rotation around its corresponding axis.
I seems the order is x->y->z.
Maybe I need some tests
Thx