lgsvl/simulator #1228: what is the variable to control  turn signal?

qwetqwe, 2021-01-26:
Hi team, I want to control turn signal of the vehicle in simulation with apollo. I set control cmd of apollo format like this:
```
    ControlCommand cmd;
    cmd.mutable_signal()->set_turn_signal(apollo::common::VehicleSignal::TURN_LEFT);
    cmd.set_left_turn(true);
    cmd.set_turnsignal(apollo::control::TURN_LEFT);
```
but the turn light is not on what I see in simulator.
how can I control turn signal？

EricBoiseLGSVL, 2021-01-26:
@qwetqwe where is this code?  In simulator?  We have VehicleStateSensor.cs that subscribes to signal topics.  Is this what you are looking for?

qwetqwe, 2021-01-28:
@EricBoiseLGSVL , the code is what I write a message in apollo to control the lgsvl vehicle signal. The topic is "/apollo/control". What is the "signal topic" you said?

qwetqwe, 2021-01-28:
I see in VehicleControlData.vs, there is no control light interface,right?
    public class VehicleControlData
    {
        // common
        public float? Acceleration; // 0..1
        public float? Braking; // 0..1

        // autoware
        public float? Velocity;
        public float? SteerAngularVelocity;
        public float? SteerAngle;
        public bool ShiftGearUp;
        public bool ShiftGearDown;

        // apollo 
        public float? SteerRate;
        public float? SteerTarget;
        public double? TimeStampSec;
        public GearPosition? CurrentGear;

        // lgsvl
        public float? SteerInput;
    }

lemketron, 2021-01-29:
@qwetqwe It looks like signal status is reported in VehicleStateSensor as @EricBoiseLGSVL mentioned above, but it does not look like the turn signal control commands from Apollo are recognized or processed in VehicleControlData (in CyberConversions.cs).

If you need this functionality and decide to implement it, please consider submitting a PR with your changes... thanks!

qwetqwe, 2021-02-01:
Got it ，thanks

closed 2021-02-01 by qwetqwe