lgsvl/simulator #1235: Unable to enable sensors: 3D Ground Truth sensor and Signal sensor

Manoj-EK, 2021-02-01:
I am currently using lgsvl simulator with Apollo stack. 
Due to problem in Apollo Perception module, I used the modular testing mentioned by lgsvl..

Here is the link to modular testing
https://youtu.be/781zkzN2xMg

My research requires me to enable and disable individual sensors..So, I started exploring quick start examples provided in PythonAPI...i came across the following code: 20-enable-sensors.py

from environs import Env
import lgsvl

env = Env()

sim = lgsvl.Simulator(env.str("LGSVL__SIMULATOR_HOST", "127.0.0.1"), env.int("LGSVL__SIMULATOR_PORT", 8181))
if sim.current_scene == "BorregasAve":
    sim.reset()
else:
    sim.load("BorregasAve")

spawns = sim.get_spawn()

state = lgsvl.AgentState()
state.transform = spawns[1]
ego = sim.add_agent(env.str("LGSVL__VEHICLE_0", "Lincoln2017MKZ (Apollo 5.0)"), lgsvl.AgentType.EGO, state)

sensors = ego.get_sensors()

for s in sensors:
    print(type(s), s.enabled)

input("Press Enter to disable lidar")

for s in sensors:
    if isinstance(s, lgsvl.LidarSensor):
        s.enabled = False

for s in sensors:
    print(type(s), s.enabled)
    
- **OUTPUT**:

[image]

I noticed that "3D Ground Truth sensor" and "Signal sensor" are not detected 

**Please do help me in this regard**

EricBoiseLGSVL, 2021-02-01:
@Manoj-EK Enable sensors with API is not working on 06 release.  Why are you needing to do this?

Manoj-EK, 2021-02-02:
@EricBoiseLGSVL  Thank you for the response....As said earlier, my research work needs me to enable and disbale individual sensors so that i can check for perception module failure behavior and apply suitable failure mitigation actoins in case of sensor failure......Is there any other alternative way that i could solve this issue?

**Please help me in this regard**

EricBoiseLGSVL, 2021-02-02:
@Manoj-EK The issue is the amount of initialization a sensor goes though on start.  This makes it very difficult to disable a sensor after.  Let me talk with the team to see if we can find a solution before the next release

Manoj-EK, 2021-02-02:
@EricBoiseLGSVL  Thank you so much....

it would be very helpful for my  further Research work if there is a way out....

I appreciate for your Help

EricBoiseLGSVL, 2021-02-02:
@Manoj-EK no problem, here to help.  This bug has been in our backlog for a bit but no users have asked about it yet.  I'll see what we can do.  If anything we can recommend a local fix for you.

Manoj-EK, 2021-02-04:
> @Manoj-EK Enable sensors with API is not working on 06 release. Why are you needing to do this?

@EricBoiseLGSVL...As I have showed in the screenshot of API script above, it can be seen that enable sensors is working but my problem was it cannot Detect "3D gound truth" sensors and "signal" sensors  in that...Is there any particular reason for this issue..
I appreciate your help in this regard

Thanks in Advance

EricBoiseLGSVL, 2021-02-05:
Does the sensor work for you before you disable?

Yimags, 2021-02-05:
@Manoj-EK  You could try to add infomation of GroundTruthSensor to [sensor.py](https://github.com/lgsvl/PythonAPI/blob/master/lgsvl/sensor.py) and [VehicleGetSensors.cs](https://github.com/lgsvl/simulator/blob/release-2020.06/Assets/Scripts/Api/Commands/VehicleGetSensors.cs).
I also edit the script a bit. And it works for me. 

```
for s in sensors:
  print(type(s), s.enabled)
input("Press Enter to disable GT")
for s in sensors:
  if s.name == "3D Ground Truth":
    s.enabled = False
for s in sensors:
  print(type(s), s.enabled)
```
[image]

Please correct me if I am wrong. @EricBoiseLGSVL I think it doesn't work before because it hasn't been defined.

Manoj-EK, 2021-02-05:
@Yimags Yes this is what was needed..
Thank you very much for information

EricBoiseLGSVL, 2021-02-05:
@Yimags Ah, yes this needs updated.  It only had a few sensors originally.