lgsvl/simulator #1236: Does Lgsvl support Nvidia GeForce RTX 2080 Mobile [8GB memory] or need high end graphics card ?

Manoj-EK, 2021-02-02:
Hi lgsvl team,

I am currently working with Apollo 6.0 using lgsvl simulation on the same system (Laptop)  

Does Lgsvl support Nvidia GeForce RTX 2080 Mobile [8GB memory] or need high end graphics card ?

EricBoiseLGSVL, 2021-02-02:
@Manoj-EK running Apollo 6.0 and simulator on the same hardware is not recommended.  We recommend using two machines connected to a gigabit switch.  Regardless, it can be done and yes lgsvl simulator will work fine on this GPU.  I use a 2070 in a laptop.  Just remember, performance depends on the resources Apollo uses and the map/sensors used.

Manoj-EK, 2021-02-02:
@EricBoiseLGSVL  Thanx for your quick response....

Manoj-EK, 2021-02-04:
@EricBoiseLGSVL  After your detailed previous response...I am planning to use two separate laptops both with Nvidia GeForce Rtx 2080 mobile [8gb memory]... One for Apollo 6.0 and another for lgsvl simulator....would that be fine to run Perception module successfully with cyber RT bridge....Thanks in Advance

lemketron, 2021-02-04:
@Manoj-EK Please note that camera perception is not yet working in Apollo 6.0. See this apollo issue: https://github.com/ApolloAuto/apollo/issues/12916#issuecomment-723741121

For now you will need to use [Modular Testing](https://www.lgsvlsimulator.com/docs/modular-testing/) to have the simulator send 3D ground truth to apollo instead of running apollo perception.

Manoj-EK, 2021-02-04:
@lemketron  thanks for your quick response.... Correct if I am wrong....I just need perception module not exclusively from Camera sensor....The documentation of Apollo (https://github.com/ApolloAuto/apollo/tree/master/modules/perception) shows Perception module= camera + lidar + radar....So, my question is....cant i manage to run perception module excluding camera sensor....or is it okay to use lower versions of Apollo like 5.5 or 5 to get perception module working....The working of Perception module is important for my research work.

And with respect to Modular Testing, it has been very much helpful in my ongoing research activity...but for my further work need Perception module from Apollo along with lgsvl simulator...Is there anyway that is feasible? 

**So please kindly do the needful**

Manoj-EK, 2021-02-20:
> @Manoj-EK running Apollo 6.0 and simulator on the same hardware is not recommended. We recommend using two machines connected to a gigabit switch. Regardless, it can be done and yes lgsvl simulator [...]

 
@EricBoiseLGSVL  @lemketron I am presently using two systems : Lgsvl simulator (WIndows10) and Apollo (ubuntu) with gigabit switch

I am finding it a bit difficult in connecting the bridge...Do I need to change config.yml file parameters? 

Is there any documentation explaining the procedure..

It would be very much helpful...

Thanks in advance

closed 2021-02-20 by Manoj-EK

mengen100, 2021-04-08:
> @lemketron thanks for your quick response.... Correct if I am wrong....I just need perception module not exclusively from Camera sensor....The documentation of Apollo [...]

@Manoj-EK Hey Manoj, I'm in the same situation right now. Have you managed to do it in any other alternative way?

Manoj-EK, 2021-04-08:
Hi,

I am presently occupied with Thesis work write-up. Will be working on it in
coming months.

lemketron, 2021-04-08:
> @lemketron thanks for your quick response.... Correct if I am wrong....I just need perception module not exclusively from Camera sensor....The documentation of Apollo [...]

You should be able to use Apollo 5.5 on the RTX-2080 but not 5.0.

With respect to specific issues related to Apollo perception please check the [Apollo project](https://github.com/ApolloAuto/apollo) documentation or [Apollo issues](https://github.com/ApolloAuto/apollo/issues) forum.