lgsvl/simulator #1239: Simulation test report

jpcloud1577, 2021-02-04:
Hi,
I just started with lgsvl and I have a basic question with regard to automated testing of AD stacks. I would like to know what is the best option to capture the simulation test results automatically when running various scripts/use cases and scenarios. For ex : I am running a vehicle following scenario with Apollo as AD stack. How to capture on simulator side if  there is some error in planning module ( may be thru callbacks?). I am looking for something like this:
1. Develop test plan ( intended behavior vs actual behavior) with various test cases
2. Run the scenario/.py file
3. Automate populating a table with test results against each of the test cases. 
Thank you for your help.

EricBoiseLGSVL, 2021-02-04:
We will be releasing an analysis feature in the next release early March.  Users can declare parameters in sensors that are captured and displayed after an iteration.  In the meantime I recommend creating a sensor that has the logic to check planning, you can write to a file or send through API (you will need to edit API also).  That way when the next release is out, you can add the data to the analysis parameters and will automatically get the callbacks in the UI.

dhdiagram4011, 2021-08-16:
Hello? Is it possible to export the simulator results performed with lgsvl cloud in file format?

EricBoiseLGSVL, 2021-08-16:
@dhdiagram4011 no you cannot export currently. What file format where you expecting? We may have this on the roadmap already for a future release and feedback/suggestions are welcome.

closed 2021-08-16 by EricBoiseLGSVL

dhdiagram4011, 2021-08-16:
yes thank you for your quick reply Finally, is it impossible to extract the result report for simulation even in the lgsvl local version? thank you.

dhdiagram4011, 2021-08-16:
Also, are there any simulation data result values ​​that the user can add and extract arbitrarily other than the test results provided in the lgsvl cloud version? thank you.

EricBoiseLGSVL, 2021-08-16:
Yes, we only provide a video if you use a video recording sensor.  We expect users to collect data that they need for comparison with the bridge or with callbacks with the api.  If you are describing a need to download the test case events, e.g. pass fail from collision, that is not supported and I am wondering what format you expect.

dhdiagram4011, 2021-08-16:
Json format or xml format
Thank you

EricBoiseLGSVL, 2021-08-18:
Excellent and thanks for the feedback.  I'll ping @thisisdavidfrank to check the roadmap for downloading analysis data.