lgsvl/simulator #1240: NPC not changing lane when forced

MartinaB85, 2021-02-04:
Good morning,

I'm using the python APIs to create a custom scenario and I have modified a little bit the script NPCLaneFollowController.cs.
In my scenario (see the attached picture) there is a NPC stopped on the left lane. Another NPC is coming behind on the same left lane. What I want to do is that when this NPC arrives behind the stopped one, it stops and, once some conditions are met, the NPC has to change lane. The conditions I am using are all met but the NPC does not change lane when forced (see a snapshot of this part of code in attached). It seems something about the speed, because if the speed of the NPC is below 3 then it can change lane, but if the speed is below 0.5 or so then it does not change lane. But I want the NPC stopping behind the stopped one, for a sec or so and then changing lane when the conditions are met.
How can I achieve my goal? 

Hope you can help me.

Thank you very much,

Martina

[image]
[image]

EricBoiseLGSVL, 2021-02-04:
Please post where you are calling this method.  It's difficult to debug with only the method.

MartinaB85, 2021-02-04:
The method is called in PhysicsUpdate()
[image]

MartinaB85, 2021-02-04:
When the NPC coming stops behind the stopped one the flag isFrontDetectWithinStopDistance and isCurve are true..so not sure if this is the issue whay is not changing lane.

MartinaB85, 2021-02-08:
Any news regarding this issue???

EricBoiseLGSVL, 2021-02-08:
IsCurve is a method to slow the vehicle when it is in a turn.  IsFrontDetectWithinStopDistance reduces the speed to avoid collision with whatever is in front of the vehicle.

Are you forcing the vehicle to change to the left lane?  The method ForceLaneChange(bool isLeft) should pass true, not false

MartinaB85, 2021-02-09:
Well, in this scenario it has to change to the right. Anyway, now that I reverted the scenario, meaning that the obstacle is on the right and the NPC has to pass to the left, it does not work properly either.
The problem is that sometimes it does change lane but some others it doesn`t and the NPC keeps staying behing the obstacle. It does enter in the function ForceLaneChange() but the NPC does nothing. 
So I would need to know why..is it because of the stopHitdistance? Is it because the car needs a specific distance from the obstacle to change lane ? I really do not understand well this behaviour..
If you could tell me please which parameters need to be checked in order for the NPC to change lane it would help.

EricBoiseLGSVL, 2021-02-10:
I'll look into this more and see if I can find a solution.

Miknow78, 2022-03-10:
> Well, in this scenario it has to change to the right. Anyway, now that I reverted the scenario, meaning that the obstacle is on the right and the NPC has to pass to the left, it does not work [...]

Hello ! Did you find your answer?
I'm with the same problem.

MartinaB85, 2022-03-10:
This was long time ago..now the code is quite different and everything seems to work fine.
Not sure what was the issue, if it was a flag that was not correctly activating or the condition on the speed. Right now I do not use any condition on the speed, only distances.

MartinaB85, 2022-03-10:
So, since you don't want to avoid collisions, if you comment if (!isFrontLeftDetect) and if (!isFrontRightDetect) in the  ForceLaneChange method the car is still not changing lane?

Miknow78, 2022-03-10:
> 

It was my mistake, sorry. 
Vehicle1 is changing lane but vehicle2 in the other lane (the lane vehicle1 is going) reduce his velocity to not collide with vehicle1. I just want them to ignore each other or remove that safe-distance to not collide.