lgsvl/simulator #1241: Error in the intersection of the HD map in customer map

qiongwu1003, 2021-02-05:
Hi,
LGSVL team

I draw my own 3D map and HD map for Apollo. And I meet some errors in the HD map when I run simulator with Apollo. I use the "Modular" mode, refer to this [vedio](https://www.youtube.com/watch?v=781zkzN2xMg&feature=youtu.be). 
I drew HD map before and I do not met the issue. Could some give me some advice?

The issue is described bellow.
In some intersections, they always show like bellow. The vehicle will never move again.
[image]
[image]

And in some intersections even can not regonize the traffic lights.
[image]

**But in some intersections do not have the issue, and they all drew in the same method as the error ones.**

The bellow is my HD map and the traffic light mesh.
[image]
[image]
[image]

Thanks.

EricBoiseLGSVL, 2021-02-06:
Any idea @lemketron ?

qiongwu1003, 2021-02-07:
I met the "Pesdestrian NavMesh" error when I build and run the simulator. Is it related to the issue above?
[image]

EricBoiseLGSVL, 2021-02-08:
No, you just can't spawn pedestrians.  You should create the NavMesh to remove this LogError.  What version of Apollo are you using?  There are issues currently with Apollo master and perception.

qiongwu1003, 2021-02-09:
@EricBoiseLGSVL Thanks for your reply.

I use the Apollo simulator, but not the latest version. And I compared the newest with mine. The "generate_map" scriptor is not changed.

EricBoiseLGSVL, 2021-02-09:
Yes, I think modular perception is only on Apollo 06.  Right @lemketron ?

lemketron, 2021-02-09:
> Yes, I think modular perception is only on Apollo 06.  Right @lemketron ?

Modular testing should work on Apollo 5.0 as well as 6.0 and master.

Traffic lights may not be detected if the exported Apollo HD map does not have signal IDs that match those in the Simulator map.

Try exporting the Apollo HD map then building the asset bundle and then saving the environment. This should preserve the generated signal IDs as long as you're on 2020.06 or later. 

You could also try assigning the Signal IDs manually before exporting and building.

EricBoiseLGSVL, 2021-02-09:
Ah, that's what I was missing.  Yes, if you manually annotate, you need to make sure all signals have signalIDs.  This should solve your issue.  Thanks @lemketron !

qiongwu1003, 2021-02-10:
@EricBoiseLGSVL @lemketron 
Thank you very much, the issue has been solved.

closed 2021-02-10 by qiongwu1003