lgsvl/simulator #1242: Some Error about Bridge plugins

hanyh1191, 2021-02-05:
When I do this topic：Docs » Plugins » Advanced » Bridge plugins
Vistual Studio2019 report: Error CS0246 could not find the type or namespace name "IBridgePlugin" and "BridgeNameAttribute" (missing using directives or assembly references?)Assembly-CSharp
And I can't  find "IBridgePlugin" and "BridgeNameAttribute" class in project. 
Why? Is there a more detailed tutorial on this topic？

EricBoiseLGSVL, 2021-02-06:
Are you on master or the 06 release branch?

hanyh1191, 2021-02-06:
> Are you on master or the 06 release branch?

Yes, I completely followed the official tutorial of LGSVL and compiled LGSVL from source code with Unity, and so far there have been no errors. 
I now make a new simulator/Asserts/External Bridges folder, and put "LoggingBridge" example clone to here, then open the project in the Unity, and error:  Assets\External\Bridges\LoggingBridge\LoggingBridgeFactory.cs(21,16): error CS0246: The type or namespace name 'IBridgeInstance' could not be found (are you missing a using directive or an assembly reference?) and several similar mistakes. And When I jump this project from Unity to Vistual Studo2019, I get an error like this:Error CS0246 could not find the type or namespace name "IBridgePlugin" and "BridgeNameAttribute" (missing using directives or assembly references?)Assembly-CSharp
Do I need to modify the LoggingBridge code? Or what should I do?