lgsvl/simulator #1243: cannot find NPC Plugins in build of simulator （NPCs vehicle build error）

1power1, 2021-02-08:
I plan to create a new NPC vehicle by myself, but when I made the prefabs and models and planned to compile, I found that there was no NPC option in the simulator's build option. This seems to be inconsistent with the official document [https://www. The description in lgsvlsimulator.com/docs/npc-plugins/#creating-npc-model-plugins] is different, how should I install this plugin.
[image]

EricBoiseLGSVL, 2021-02-08:
Please screen shot your NPC folders.

1power1, 2021-02-09:
> Please screen shot your NPC folders.
The content contained in the NPC folder is as shown below. I added a prefab model of an NPC vehicle in simulator/Assets/External/NPCs. In order to verify whether it is a problem with the model made by myself, I found it in simulator/Assets/Models/ Under the NPC folder, only the Mesh of the body of the original model has been modified, and the model name remains unchanged. The same operation was performed in the simulator/Assets/Prefabs/NPC folder. Finally, the NPCManager under the simulator/Assets/Prefabs/Managers folder is shown in the figure. However, the NPC vehicles in the directly generated map have not changed.
[image]
[image]
[image]
[image]

EricBoiseLGSVL, 2021-02-09:
If you add NPC models to the NPCManager and the NPCSettings in Resources like so
[image]

Then you will get NPC included when you create a new binary.

If you want to create NPC bundles you need to add to the External folder like so
[image]

This will create a bundle that must be placed in the AssetBundles/NPCs folder so the binary or source in editor can load on start.  Then you can place in scene with API or randomly in random traffic.  This method is not fully supported yet and we are working to include them in the options when you create a simulation.

1power1, 2021-02-09:
I still don’t know how to do it, but I have replaced the model in NPCManager and the NPCSettings, but it still doesn’t work when recompiling the map file. As for the second method, I’m in simulator/AssetBundles/ A new folder was created under NPCs, and the corresponding NPC model was put into it, but when Unity was opened, there was no such folder under simulator/Assets/External/NPCs (not even after restarting Unity). You can start from the beginning (no Explain the status of any changes made to the project? Or record a video, thank you very much!

EricBoiseLGSVL, 2021-02-09:
1. Put the edited mesh in the NPC/Models folder
2. Create a new prefab for the NPC, RENAME it
3. Place it in Prefabs/NPC
4. Edit NPCManager and add the new NPC prefab
5. Edit NPCSettings and add the new NPC prefab
6. Create a new binary with your changes.  No need to build a bundle, the asset is in the project now.  Recompiling and map files have nothing to do with this.

The External Bus is just an example.  You won't have this.  You would need to create a new repo.  That is what the External folder is for.  External repos that you clone to edit.  Since you don't have any example repos to use, don't worry about it.

Be sure all settings and scripts are correct on the new NPC you created.  Look to the other NPC as an example.

1power1, 2021-02-22:
> 1. Put the edited mesh in the NPC/Models folder
> 2. Create a new prefab for the NPC, RENAME it
> 3. Place it in Prefabs/NPC
> 4. Edit NPCManager and add the new NPC prefab
> 5. Edit NPCSettings and add the [...]

I am now sure that I have performed the first five steps correctly according to the first five steps you gave. The problem may appear in the last step "Create a new binary with my changes". What format is the file I need to generate? Can you explain this step in detail below? thank you very much!

lemketron, 2021-02-22:
> I am now sure that I have performed the first five steps correctly according to the first five steps you gave. The problem may appear in the last step "Create a new binary with my changes". What format is the file I need to generate? Can you explain this step in detail below? thank you very much!

I believe @EricBoiseLGSVL was suggesting that you build a new Simulator executable binary, as described in [build-instructions](https://www.lgsvlsimulator.com/docs/build-instructions/).

rogeherlor, 2021-04-03:
> 1. Put the edited mesh in the NPC/Models folder
> 2. Create a new prefab for the NPC, RENAME it
> 3. Place it in Prefabs/NPC
> 4. Edit NPCManager and add the new NPC prefab
> 5. Edit NPCSettings and add the [...]

Does this work for adding a new NPC?
I am trying to create a new Maxivan adding it like the other NPCs models and prefabs
[image]
Editing the NPCManager and NPCSettings
[image]
[image]
But when I run the simulation it disappears
[image]

EricBoiseLGSVL, 2021-04-05:
Did you save the changes to the prefab and rebuild the simulator binary?  FYI, soon we will release npc bundles that can be created from the build menu :)

rogeherlor, 2021-05-08:
> Did you save the changes to the NPCManager prefab and rebuild the simulator binary? Also, be sure MapOrigin in CubeTown includes the MiniVan type (it does but for anyone else looking at this). FYI, soon we will release npc and pedestrian bundles that can be created from the build menu. :)

Yes, I did, but I still can't include my NPCs. I have tried editing the existing SUV model, but a simple little cube makes the model look strange as in the second picture. I built it too and the SUV spawns and it is destroyed.

[image]

[image]

When will the NPC build menu release be available? It would be great

EricBoiseLGSVL, 2021-05-11:
Spawns and destroyed? What does this mean?  Are you getting errors?  It looks like you didn't zero the holder objects when you duplicated the wheel collider holder.  Why are you adding a cube?  What are you trying to do with the SUV?