lgsvl/simulator #1246: Change Camera Views Using Python Apis

abhkri, 2021-02-10:
Hi

Currently, there are 3 camera views 'Follow', 'Cinematic' and 'Free' available in LGSVL simulator. How to perform the following activities while running simulator from Python Api's:
1) Switch between camera views, 
2) Control zoom in/out of a particular camera angle,
3) Offset position of camera to either left or right direction to get better view of scenario.

EricBoiseLGSVL, 2021-02-12:
We have an update to camera control in the next release.  We have API control for FreeMode and you can set the position and rotation.  From this you can create your own control modes you need.  This will release very soon

NamanJhamb, 2021-04-14:
I think you are referring to 34-simulator-cam-set.py over here. Is it also possible to set up the "Follow" camera inside the vehicle to get a driver's perspective of the simulation? Also, can I find a reference for the coordinates of different maps (especially SanFrancisco)? Thanks!

> We have an update to camera control in the next release. We have API control for FreeMode and you can set the position and rotation. From this you can create your own control modes you need. This will release very soon

EricBoiseLGSVL, 2021-04-14:
@NamanJhamb yes but not through API at this time.  We have the UI button that changes to driver camera.  It is outside the vehicle at the moment.  You can edit the vehicle and place a public transform for a different view position.  Keep in mind our vehicles don't have interiors.  Unity scenes are use the unity coordinates.  You can run the simulator in the editor and see the positions in the inspector.  You can create a sensor to output these coordinates as well.

rajeshjoshi8792, 2021-06-25:
@EricBoiseLGSVL is changes to driver camera  through API is implemented?

EricBoiseLGSVL, 2021-06-26:
We have added the ability to set the simulator camera state in the next release.  The other features are a very narrow use case and you will need to implement

closed 2021-06-26 by EricBoiseLGSVL

rajeshjoshi8792, 2021-06-28:
@EricBoiseLGSVL is there any expected date for next release? i have to implement simulator camera as a driver perspective. so is there any help that you can do to implement this using API?

EricBoiseLGSVL, 2021-06-28:
We have the feature in review and we are running QA on the release right now.  We hope by the end of the week.  If you can't wait:

simulator new api command class
```
/**
 * Copyright (c) 2021 LG Electronics, Inc.
 *
 * This software contains code licensed as described in LICENSE.
 *
 */

using SimpleJSON;
using System;
using UnityEngine;

namespace Simulator.Api.Commands
{
    class SimulatorCameraStateSet : IDistributedCommand
    {
        public string Name => "simulator/camera/state/set";

        public void Execute(JSONNode args)
        {
            var camManager = SimulatorManager.Instance.CameraManager;
            var api = ApiManager.Instance;

            if (camManager == null)
            {
                api.SendError(this, "Camera Manager not found. Is the scene loaded?");
                return;
            }

            var state = args["state"].AsInt;
            camManager.SetCameraState((CameraStateType)state);
            api.SendResult(this);
        }
    }
}

```

and python edits simulator.py
```
14. from enum import Enum

30. class SimulatorCameraState(Enum):
          FREE = 0
          FOLLOW = 1
          CINEMATIC = 2
          DRIVER = 3

90. @accepts(SimulatorCameraState)
      def set_sim_camera_state(self, state):
          self.remote.command("simulator/camera/state/set", {"state": state.value})
```

rajeshjoshi8792, 2021-06-30:
hello team,

Thank you for your response.

i'm implementing this code. and getting one error.

"Assets\Scripts\Api\Commands\SimulatorCameraStateSet.cs(30,24): error
CS1061: 'CameraManager' does not contain a definition for 'SetCameraState'
and no accessible extension method 'SetCameraState' accepting a first
argument of type 'CameraManager' could be found (are you missing a using
directive or an assembly reference?)"

I checked CameraManager class. but did not find any SetCameraState method.

please help me to solve this.

Thank You

EricBoiseLGSVL, 2021-06-30:
Sorry, missing that method.  I edited the previous post :)

rajeshjoshi8792, 2021-07-01:
Thank you for your reply. 
i am able to implement code. but after ininitalizing camera state, i m getting below error.

Traceback (most recent call last):
  File "twocaraccident.py", line 23, in <module>
    sim.set_sim_camera_state(sim.SimulatorCameraState.DRIVER)
  File "C:\Users\rjjos\AppData\Roaming\Python\Python38\site-packages\lgsvl\utils.py", line 23, in new_f
    return f(*args, **kwargs)
  File "C:\Users\rjjos\AppData\Roaming\Python\Python38\site-packages\lgsvl\simulator.py", line 105, in set_sim_camera_state
    self.remote.command("simulator/camera/state/set", {"state": state.value})
  File "C:\Users\rjjos\AppData\Roaming\Python\Python38\site-packages\lgsvl\remote.py", line 73, in command
    raise Exception(data["error"])
Exception: Object reference not set to an instance of an object.

i inintialize camera state like this. 
sim.set_sim_camera_state(sim.SimulatorCameraState.DRIVER)

rajeshjoshi8792, 2021-07-02:
> Thank you for your reply.
> i am able to implement code. but after ininitalizing camera state, i m getting below error.
> 
> Traceback (most recent call last):
> File "twocaraccident.py", line 23, in
> sim.set_sim_camera_state(sim.SimulatorCameraState.DRIVER)
> File "C:\Users\rjjos\AppData\Roaming\Python\Python38\site-packages\lgsvl\utils.py", line 23, in new_f
> return f(*args, **kwargs)
> File "C:\Users\rjjos\AppData\Roaming\Python\Python38\site-packages\lgsvl\simulator.py", line 105, in set_sim_camera_state
> self.remote.command("simulator/camera/state/set", {"state": state.value})
> File "C:\Users\rjjos\AppData\Roaming\Python\Python38\site-packages\lgsvl\remote.py", line 73, in command
> raise Exception(data["error"])
> Exception: Object reference not set to an instance of an object.
> 
> i inintialize camera state like this.
> sim.set_sim_camera_state(sim.SimulatorCameraState.DRIVER)

can you help me to resolve this issue?

EricBoiseLGSVL, 2021-07-02:
this is working as intended.  I am not sure what you have written in twocaraccident.py to get this error.
```
sim.set_sim_camera_state(sim.SimulatorCameraState.DRIVER)
```
this works and I have given you all the code in the next release for this feature. You'll need to post your code to understand why you are getting an error.

rajeshjoshi8792, 2021-07-02:
this is the code for twocaraccident.py

from environs import Env
import lgsvl

env = Env()

sim = lgsvl.Simulator(env.str("LGSVL__SIMULATOR_HOST", lgsvl.wise.SimulatorSettings.simulator_host), env.int("LGSVL__SIMULATOR_PORT", lgsvl.wise.SimulatorSettings.simulator_port))
if sim.current_scene == lgsvl.wise.DefaultAssets.map_borregasave:
    sim.reset()
else:
    sim.load(lgsvl.wise.DefaultAssets.map_borregasave)

spawns = sim.get_spawn()
forward = lgsvl.utils.transform_to_forward(spawns[0])
right = lgsvl.utils.transform_to_right(spawns[0])

# ego vehicle
state = lgsvl.AgentState()
state.transform = spawns[0]
state.velocity = 20 * forward
ego = sim.add_agent(env.str("LGSVL__VEHICLE_0", lgsvl.wise.DefaultAssets.ego_jaguar2015xe_apollo5), lgsvl.AgentType.EGO, state)

sim.set_sim_camera_state(sim.SimulatorCameraState.DRIVER)

vehicles = {
    ego: "EGO",
    # bus: "SchoolBus",
    # sedan: "Sedan",
}

# This function gets called whenever any of the 3 vehicles above collides with anything
def on_collision(agent1, agent2, contact):
    name1 = vehicles[agent1]
    name2 = vehicles[agent2] if agent2 is not None else "OBSTACLE"
    print("{} collided with {} at {}".format(name1, name2, contact))

ego.on_collision(on_collision)
sim.run(10)

EricBoiseLGSVL, 2021-07-02:
code looks fine, please post the simulator.py changes you made

rajeshjoshi8792, 2021-07-02:
https://drive.google.com/file/d/1wJyHmCKEpCDMkrVpE-GwX7wwGtMsa6aQ/view?usp=sharing

EricBoiseLGSVL, 2021-07-02:
just post here please

rajeshjoshi8792, 2021-07-02:
#
# Copyright (c) 2019-2021 LG Electronics, Inc.
#
# This software contains code licensed as described in LICENSE.
#

from .remote import Remote
from .agent import Agent, AgentType, AgentState
from .sensor import GpsData
from .geometry import Vector, Transform, Spawn
from .utils import accepts, ObjectState
from .controllable import Controllable
from enum import Enum
from collections import namedtuple
from environs import Env
from datetime import datetime
import re

RaycastHit = namedtuple("RaycastHit", "distance point normal")

WeatherState = namedtuple("WeatherState", "rain fog wetness cloudiness damage")
WeatherState.__new__.__defaults__ = (0,) * len(WeatherState._fields)

env = Env()

class Simulator:

    class SimulatorCameraState(Enum):
        FREE = 0
        FOLLOW = 1
        CINEMATIC = 2
        DRIVER = 3

    @accepts(str, int)
    def __init__(self, address=env.str("LGSVL__SIMULATOR_HOST", "localhost"), port=env.int("LGSVL__SIMULATOR_PORT", 8181)):
        if port <= 0 or port > 65535:
            raise ValueError("port value is out of range")
        self.remote = Remote(address, port)
        self.agents = {}
        self.callbacks = {}
        self.stopped = False

    def close(self):
        self.remote.close()

    @accepts(str, int)
    def load(self, scene, seed=None):
        self.remote.command("simulator/load_scene", {"scene": scene, "seed": seed})
        self.agents.clear()
        self.callbacks.clear()

    @property
    def version(self):
        return self.remote.command("simulator/version")

    @property
    def layers(self):
        return self.remote.command("simulator/layers/get")

    @property
    def current_scene(self):
        return self.remote.command("simulator/current_scene")

    @property
    def current_scene_id(self):
        return self.remote.command("simulator/current_scene_id")

    @property
    def current_frame(self):
        return self.remote.command("simulator/current_frame")
[... 254 lines cut ...]
    def controllable_remove(self, controllable):
        self.remote.command("simulator/controllable_remove", {"uid": controllable.uid})
        del self.controllables[controllable.uid]

    @accepts(str)
    def get_controllables(self, control_type=None):
        j = self.remote.command("controllable/get/all", {
            "type": control_type,
        })
        return [Controllable(self.remote, controllable) for controllable in j]

    @accepts(str)
    def get_controllable_by_uid(self, uid):
        j = self.remote.command("controllable/get", {
            "uid": uid,
        })
        return Controllable(self.remote, j)

    @accepts(Vector, str)
    def get_controllable(self, position, control_type=None):
        j = self.remote.command("controllable/get", {
            "position": position.to_json(),
            "type": control_type,
        })
        return Controllable(self.remote, j)

EricBoiseLGSVL, 2021-07-02:
Why do you have a print in the setter? Other than that it is fine.  What version of simulator are you using?  Are you using APIOnly simulation?  This will not work with PythonAPI test case, APIOnly since you are changing the API source

rajeshjoshi8792, 2021-07-02:
> Why do you have a print in the setter? Other than that it is fine. What version of simulator are you using? Are you using APIOnly simulation? This will not work with PythonAPI test case, APIOnly since you are changing the API source

1.Why do you have a print in the setter?
ans : i just want to check that if i m getting state value or not.

2. What version of simulator are you using?
ans : 2021.1

3.  Are you using APIOnly simulation?
ans : yes

EricBoiseLGSVL, 2021-07-02:
Did you get the value? So the simulation type in webui is ApiOnly, not PythonAPI? Like so:
[image]

rajeshjoshi8792, 2021-07-02:
yes i m getting value as 3. and yes after s selecting imulation menu, i clicked on api only simulator.

EricBoiseLGSVL, 2021-07-02:
The only thing I can think of is did you install python with this?
```
pip install -e .
```

otherwise changes might not be used

rajeshjoshi8792, 2021-07-02:
yes i install same way. but i think the issue is in api call. api not hitting SimulatorCameraStateSet file.

EricBoiseLGSVL, 2021-07-02:
Are you creating a new simulator binary because you are adding source code?

rajeshjoshi8792, 2021-07-02:
yes, i m created new binary. and executing that binary to opening simulator.

rajeshjoshi8792, 2021-07-02:
is it possible for you to send me latest binary that include this feauture?

EricBoiseLGSVL, 2021-07-02:
no we cannot do that.  What errors are you getting in simulator (not just python?)  please post your player.log

rajeshjoshi8792, 2021-07-02:
NullReferenceException: Object reference not set to an instance of an object
  at SimulatorCameraController.SetDriverViewCameraState () [0x00032] in <e742d4c180e147c99677010334153530>:0 
  at CameraManager.SetCameraState (CameraStateType state) [0x0004a] in <e742d4c180e147c99677010334153530>:0 
  at Simulator.Api.Commands.SimulatorCameraStateSet.Execute (SimpleJSON.JSONNode args) [0x00038] in <e742d4c180e147c99677010334153530>:0 
  at Simulator.Api.ApiManager.Update () [0x0012d] in <e742d4c180e147c99677010334153530>:0 
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
Simulator.Api.ApiManager:Update()
 
(Filename: <e742d4c180e147c99677010334153530> Line: 0)

02-07-2021 23:40:08|Fatal|<>c__DisplayClass174_0.<startReceiving>b__2:0|System.IO.IOException: Unable to read data from the transport connection: An existing connection was forcibly closed by the remote host.
                          . ---> System.Net.Sockets.SocketException: An existing connection was forcibly closed by the remote host.
                          
                            at System.Net.Sockets.Socket.EndReceive (System.IAsyncResult asyncResult) [0x00012] in <ef151b6abb5d474cb2c1cb8906a8b5a4>:0 
                            at System.Net.Sockets.NetworkStream.EndRead (System.IAsyncResult asyncResult) [0x00057] in <ef151b6abb5d474cb2c1cb8906a8b5a4>:0 
                             --- End of inner exception stack trace ---
                            at System.Net.Sockets.NetworkStream.EndRead (System.IAsyncResult asyncResult) [0x0009b] in <ef151b6abb5d474cb2c1cb8906a8b5a4>:0 
                            at WebSocketSharp.Ext+<>c__DisplayClass59_0.<ReadBytesAsync>b__0 (System.IAsyncResult ar) [0x00002] in <84d0020558774b6c88b3203b756a09ec>:0  
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

EricBoiseLGSVL, 2021-07-02:
you have a bug in simulator code.  That is why the error is happening.  Please show me the code in the api command code and in the CameraManager set camera state

rajeshjoshi8792, 2021-07-02:
CameraManager .cs

using UnityEngine;

public class CameraManager : MonoBehaviour
{
    public GameObject SimulatorCameraPrefab;
    public Camera SimulatorCamera { get; private set; }
    private SimulatorCameraController CameraController;
    
    private void Awake()
    {
        SimulatorCamera = Instantiate(SimulatorCameraPrefab, transform).GetComponentInChildren<Camera>();
        CameraController = SimulatorCamera.GetComponentInParent<SimulatorCameraController>();
    }

    private void OnEnable()
    {
        SimulatorManager.Instance.AgentManager.AgentChanged += OnAgentChange;
    }

    private void OnDisable()
    {
        SimulatorManager.Instance.AgentManager.AgentChanged -= OnAgentChange;
    }

    private void OnAgentChange(GameObject agent)
    {
        CameraController.SetFollowCameraState(agent);
    }

    public void SetFreeCameraState()
    {
        CameraController.SetFreeCameraState();
    }

    /// <summary>
    /// API command to set free camera position and rotation
    /// </summary>
    public void SetFreeCameraState(Vector3 pos, Vector3 rot)
    {
        CameraController.SetFreeCameraState(pos, rot);
    }

    // <summary>
    /// API command to set camera state
    /// </summary>
    public void SetCameraState(CameraStateType state)
    {
        switch (state)
        {
            case CameraStateType.Free:
                CameraController.SetFreeCameraState();
                break;
            case CameraStateType.Follow:
                CameraController.SetFollowCameraState(SimulatorManager.Instance.AgentManager.CurrentActiveAgent);
                break;
            case CameraStateType.Cinematic:
                CameraController.SetCinematicCameraState();
                break;
            case CameraStateType.Driver:
                CameraController.SetDriverViewCameraState();
                break;
        }
    }
    public void ToggleCameraState()
    {
        CameraController.IncrementCameraState();
        switch (CameraController.CurrentCameraState)
        {
            case CameraStateType.Free:
                CameraController.SetFreeCameraState();
                break;
            case CameraStateType.Follow:
                CameraController.SetFollowCameraState(SimulatorManager.Instance.AgentManager.CurrentActiveAgent);
                break;
            case CameraStateType.Cinematic:
                CameraController.SetCinematicCameraState();
                break;
        }
        SimulatorManager.Instance.UIManager?.SetCameraButtonState();
    }

    public CameraStateType GetCurrentCameraState()
    {
        return CameraController.CurrentCameraState;
    }

    public void Reset()
    {
        CameraController.SetFollowCameraState(gameObject);
    }
}

rajeshjoshi8792, 2021-07-02:
SimulatorCameraStateSet .cs

using SimpleJSON;
using System;
using UnityEngine;

namespace Simulator.Api.Commands
{
    class SimulatorCameraStateSet : IDistributedCommand
    {
        public string Name => "simulator/camera/state/set";

        public void Execute(JSONNode args)
        {
            var camManager = SimulatorManager.Instance.CameraManager;
            var api = ApiManager.Instance;

            if (camManager == null)
            {
                api.SendError(this, "Camera Manager not found. Is the scene loaded?");
                return;
            }

            var state = args["state"].AsInt;
            camManager.SetCameraState((CameraStateType)state);
            api.SendResult(this);
        }
    }
}

EricBoiseLGSVL, 2021-07-02:
Do you have any compile errors in the code you changed in simulator?

rajeshjoshi8792, 2021-07-02:
no. i can able to build simulator without any error.

EricBoiseLGSVL, 2021-07-02:
Do you have this method in simulatorcameracontrol.cs?
```
line 410 SetDriverViewCameraState()
```

Does your vehicle have a driver view object?  Can you click in the bottom right in the UI to manually go to driver view in a random simulation with this same vehicle?  The stack trace says it has a null in this method.  The only object it references is this driver view transform. This is the error you are having.

EricBoiseLGSVL, 2021-07-02:
what version of simulator are you working on? master? tag release-2021.1?

rajeshjoshi8792, 2021-07-02:
> Do you have this method in simulatorcameracontrol.cs?
>
> ```
> line 410 SetDriverViewCameraState()
> ```
>
> Does your vehicle have a driver view object? Can you click in the bottom right in the UI to [...]

public void SetDriverViewCameraState()
    {
        SimulatorManager.Instance?.UIManager?.ResetCinematicAlpha();
        CurrentCameraState = CameraStateType.Driver;
        targetObject = SimulatorManager.Instance.AgentManager.CurrentActiveAgent.transform;
        DriverViewCameraTransform = targetObject.GetComponent<IAgentController>().DriverViewTransform;
        if (DriverViewCameraTransform != null)
        {
            transform.SetParent(DriverViewCameraTransform);
            transform.localPosition = Vector3.zero;
            transform.localRotation = Quaternion.identity;
            thisCamera.transform.localRotation = Quaternion.identity;
            thisCamera.transform.localPosition = Vector3.zero;
        }
        SimulatorManager.Instance.UIManager?.SetCameraButtonState();
    }

rajeshjoshi8792, 2021-07-02:
> tag release-2021.1

tag release-2021.1

rajeshjoshi8792, 2021-07-02:
> vehicle have a driver view object

yes

EricBoiseLGSVL, 2021-07-02:
can you run a normal simulation with the vehicle you are trying to use and get the driver view to work?

rajeshjoshi8792, 2021-07-02:
> can you run a normal simulation with the vehicle you are trying to use and get the driver view to work?

yes i can able to use driver view using UI.

EricBoiseLGSVL, 2021-07-02:
I recommend you attach the visual studio debugger to see what is null in your scripts.  Start with the null stack trace at the point in the player.log specified.  Then look at the values in the debugger.  Worst case, this feature will release very soon.

rajeshjoshi8792, 2021-07-02:
is there any expected date for release? i need asap

EricBoiseLGSVL, 2021-07-02:
we are going through QA now, should be in the next week or the week after.  I would test other api quick scripts, debug the null you are getting.  The issue may be a difference in code from master to release that can't be fixed until the release.  You could try these changes on master of github just be aware you must use preview webui assets.

rajeshjoshi8792, 2021-07-04:
i build simulator as a development mode. and get the exact error.

NullReferenceException: Object reference not set to an instance of an object
  at SimulatorCameraController.SetDriverViewCameraState () [0x00024] in D:\python\SLV Simulator\SVL 2021.1\simulator\Assets\Scripts\Controllers\SimulatorCameraController.cs:414 
  at CameraManager.SetCameraState (CameraStateType state) [0x00054] in D:\python\SLV Simulator\SVL 2021.1\simulator\Assets\Scripts\Managers\CameraManager.cs:67 
  at Simulator.Api.Commands.SimulatorCameraStateSet.Execute (SimpleJSON.JSONNode args) [0x0003e] in D:\python\SLV Simulator\SVL 2021.1\simulator\Assets\Scripts\Api\Commands\SimulatorCameraStateSet.cs:30 
  at Simulator.Api.ApiManager.Update () [0x0014c] in D:\python\SLV Simulator\SVL 2021.1\simulator\Assets\Scripts\Api\ApiManager.cs:498 
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
Simulator.Api.ApiManager:Update() (at D:\python\SLV Simulator\SVL 2021.1\simulator\Assets\Scripts\Api\ApiManager.cs:516)
 
(Filename: D:/python/SLV Simulator/SVL 2021.1/simulator/Assets/Scripts/Controllers/SimulatorCameraController.cs Line: 414)

can you help me in that?

rajeshjoshi8792, 2021-07-05:
I think im close to implement this. If you can help me to solve that it will be easy for me. Please help me to solve this issue.

EricBoiseLGSVL, 2021-07-06:
I think the vehicle you are using doesn't have the driver view transform.  Please create a new vehicle on webui with a vehicle you are sure has a driver view transform and replace the sensor config id with the default one in API.  Then see if you get the same error.

rajeshjoshi8792, 2021-07-07:
> I think the vehicle you are using doesn't have the driver view transform. Please create a new vehicle on webui with a vehicle you are sure has a driver view transform and replace the sensor config id with the default one in API. Then see if you get the same error.

Can you confirm me if SetDriverViewCameraState() method is correct in version 2020.2? Because i copy that method from version 2020.2. and also add some class that not available in version 2020.1 in this method.

rajeshjoshi8792, 2021-07-07:
And also i try to run version 2020.2, but it always give me "outofdate map bundle error".

EricBoiseLGSVL, 2021-07-07:
We haven't released version 2020.2.  Where are you seeing this? SetDriverViewCameraState is correct in latest release (not out yet).  What class did you add?

rajeshjoshi8792, 2021-07-07:
i checkout branch release-2021.2 from this current repo.  i added IAgentController, ISensorsController class.

and also please guide me where i changed this url in code?
https://wise.preview.svlsimulator.com

EricBoiseLGSVL, 2021-07-07:
in the config.yml file in root of source code/binary
Sorry, I can't debug code I can't see.  Unless you post all changes from source I have no idea why you are getting errors.  If you run 2021.2 branch with all new assets, without your changes, do you get the correct functionality?  Try this first and then open a new issue for help with your added code.  This issue has been closed and we aren't helping other users.

rajeshjoshi8792, 2021-07-07:
> I think the vehicle you are using doesn't have the driver view transform. Please create a new vehicle on webui with a vehicle you are sure has a driver view transform and replace the sensor config id with the default one in API. Then see if you get the same error.

[image]

i m trying to add vehicle uisng Ui. but i m struck in this screen. where can i find asset bundle?

EricBoiseLGSVL, 2021-07-07:
In the source root AssetBundles folder

rajeshjoshi8792, 2021-07-07:
> In the source root AssetBundles folder

i checked AssetBundles in the root. and able to see some subfolder like environment, Vehicles etc. but all that sub folder are empty. so how i add asset bundle in UI?

EricBoiseLGSVL, 2021-07-07:
If you don't have a asset bundle in the simulator source root, then it did not build.  You have to build the asset in the Build menu in the Editor.  Please see doc [here](https://www.svlsimulator.com/docs/simulation-content/add-new-ego-vehicle/).

rajeshjoshi8792, 2021-07-07:
> in the config.yml file in root of source code/binary
> Sorry, I can't debug code I can't see. Unless you post all changes from source I have no idea why you are getting errors. If you run 2021.2 [...]

i can see config.yml.template file. this code are exist in this file
headless: false
read_only: false
api_hostname: "localhost"
api_port: 8181
cloud_url: "https://wise.svlsimulator.com"

so i changed cloud_url to this.
cloud_url: "https://wise.preview.svlsimulator.com".

and create buill. but it still goes to https://wise.svlsimulator.com

EricBoiseLGSVL, 2021-07-07:
sorry you also need to delete the appdata folder because it is defaulting to wise.sim.  ```C:\Users\USER\AppData\LocalLow\LGElectronics\SVLSimulator```
Make sure no webui is open and connected when you are re-linking to webui preview.sim

rajeshjoshi8792, 2021-07-07:
> sorry you also need to delete the appdata folder because it is defaulting to wise.sim. `C:\Users\USER\AppData\LocalLow\LGElectronics\SVLSimulator`
> Make sure no webui is open and connected when you are re-linking to webui preview.sim

yes, now i m redirecting preview.sim. and map also working fine. but when i creating ego vehicle i m getting "value can not be null" error. below is the log

Traceback (most recent call last):
  File "twocaraccidenttest.py", line 25, in <module>
    ego = sim.add_agent(env.str("LGSVL__VEHICLE_0", lgsvl.wise.DefaultAssets.ego_jaguar2015xe_apollo5), lgsvl.AgentType.EGO, state)
  File "C:\Users\rjjos\AppData\Roaming\Python\Python38\site-packages\lgsvl\utils.py", line 23, in new_f
    return f(*args, **kwargs)
  File "C:\Users\rjjos\AppData\Roaming\Python\Python38\site-packages\lgsvl\simulator.py", line 167, in add_agent
    uid = self.remote.command("simulator/add_agent", args)
  File "C:\Users\rjjos\AppData\Roaming\Python\Python38\site-packages\lgsvl\remote.py", line 73, in command
    raise Exception(data["error"])
Exception: Value cannot be null.
Parameter name: path2

EricBoiseLGSVL, 2021-07-07:
You need to paste the sensor config of the new vehicle that has a driver view and is rebuilt for the new version here ```lgsvl.wise.DefaultAssets.ego_jaguar2015xe_apollo5```.  It is a guid that you click on in the sensor config view.  This loads the vehicle and the sensors associated with it.  The one you are referencing now is not updated for preview.

rajeshjoshi8792, 2021-07-07:
> You need to paste the sensor config of the new vehicle that has a driver view and is rebuilt for the new version here `"lgsvl.wise.DefaultAssets.ego_jaguar2015xe_apollo5"`. (put the guid in "quotes") It is a guid that you click on in the sensor config view. This loads the vehicle and the sensors associated with it. The one you are referencing now is not updated for preview.

i m not  clearly this. can you explain me how i do this?

[13 more replies]