lgsvl/simulator #1248: Working with maps of real cities

PavelLysenok, 2021-02-12:
Hi!

I tried to edit/make the maps of two small territories (less than 1x1 km each) based on pointcloud data using LGSVL simulator, but I've faced several problems and have some questions:

1. The most important thing - Unity runs too slow when I import a .xodr map of the first area. My laptop starts acting up when the real world data (with just about 2800 "MapLaneSection_road"s - yeah I know that even the biggest open source OpenDrive maps from the LGSVL website (for example San Francisco) have less, but the real world is too complicated to be described by just straight lines) is uploaded. At the same time the .ply file has more than 9 000 000 points - and everything goes okay just with them uploaded. And I understand that the second area will be even bigger than the first one. Is it okay that I face this problem and LGSVL simulator is not intended for creating real world maps? Or are there any solutions for this?

2. Are there any ways to transform an already existing boundary line into a lane?

3. Is it possible to draw waypoints using the cursor and placing them just on the exact point on the pointcloud substrate?

4. The program always crashes when I try to change the size of the hierarchy window, the project window etc. 

Ubuntu 18.04
memory: 16 GB
processor: Intel® Core™ i7-10510U CPU @ 1.80GHz × 8
graphics: GeForce MX250/PCIe/SSE2

EricBoiseLGSVL, 2021-02-12:
@PavelLysenok Hello!

1) Minimum recommended GPU is a 1080 with 8 gb memory.  I'm surprised you can run such a large point cloud simulation with this laptop GPU at all.  We support real world maps but we also specify minimum requirements.  Memory requirements increase as you increase the size of the map/textures/points/lights etc.
2) What do you mean exactly?  MapLane and MapLine are different annotation classes.  Open the annotation tool in Unity Editor and select the MapLine/MapLane and see the components attached to the annotation object in scene.  Why would you swap this?  Wouldn't this not match the environment then?  More details on what you are trying to do will help answer this better.
3) Yes, but we do not support this currently.  You would need to capture mouse click and screen pos, then calculate real world position from screen space or raycast to the point in world coordinates.  We have plans to move the annotation tool out of Editor and into the binary but this is just starting development and won't be ready for a while.
4) This is a known bug with Linux and Unity.  We are hoping to upgrade Unity soon to pull back ported fixes from unity into the 2019.4 LTS but it is still being tested.  Set default layout in editor and be sure all windows are docked.

PavelLysenok, 2021-02-15:
@EricBoiseLGSVL thank you for your answer!

1. To be honest, I was also surprised that even such a big pointcloud doesn't affect the efficiency much, but the problems start only with lane-data (even if the pointcloud is removed from the scene)

2. I have an OpenDrive format map file, but it has problems with real lanes locations - the "real lanes" are represented as MapLines and MapLanes are situated somewhere to the left from the original position (it's shown on the screenshots below). So my idea was to delete all excess object manually and to convert the MapLines (with the correct position) to MapLanes with default parameters - and to rewrite the attributes (speed limits, lane turns, boundary lines etc) manually afterwards.

[image]

[image]

Maybe it'll be better to redraw the map manually, but due to the problem I mentioned above, it seems to be impossible due to the speed of work. And, unfortunately, I'm pretty much limited in time. 

3-4. Great! Will be waiting for the upcoming updates. :)

EricBoiseLGSVL, 2021-02-16:
1) yes lane data is very heavy.  Unity Editor has a different update cycle and has issues with large annotations.  We are looking into better was to mitigate this.

2) The real world and the annotation in scene looks very different.  Is the OpenDrive map from a real location?  The new release will fix the duplicated data in the inspector and may help with importing.  You can manually  annotate anytime and since this is so off I would recommend it.  You can reduce the annotation and simplify for the simulation.  NPCs will have issues with this many annotation points.

3-4) Yes, we can't wait to release all the new updates and fixes :)

PavelLysenok, 2021-02-18:
@EricBoiseLGSVL 

1-2. Yes, it's a map from a real location - that's why I care so much about the accuracy of the map construction and that's why I have so many waypoints. So, the only way to build a map is to draw it manually, I guess.

Thank you again for the answers!

EricBoiseLGSVL, 2021-02-18:
@PavelLysenok I think the new release will improve the import of the hdmap.  It is odd that the curve is so off and maplines are where maplanes should be.  Yes, I recommend manually annotating.  You should also have intersections were the traffic lanes meet.  Let's keep this open until you can try the importer with the new release.  Sorry that you are having issues.