lgsvl/simulator #1251: As an autonomous vehicles researcher, I need to know the simulation's ground-truth state, so that I can develop/debug autonomous driving algorithms.

TrentWeiss, 2021-02-12:
I see that it's possible to have the simulator publish the [ground-truth states](https://github.com/lgsvl/lgsvl_msgs/blob/foxy-devel/msg/Detection3DArray.msg) of any other traffic agents (both NPC agents and other agents driven by whatever bridge).

Is it possible to infer the ground-truth simulated state of the _ego vehicle_. I.e. the pose/velocity of the ego expressed in the sim's global coordinate system.  I also see that if you configure [a gps odometry sensor](https://www.lgsvlsimulator.com/docs/sensor-json-options/#gps-odometry) on the ego, it manifests as a ROS2 topic of type [Odometry](https://github.com/ros2/common_interfaces/blob/foxy/nav_msgs/msg/Odometry.msg) on the other side of the ROS2 bridge.  Are the values in the Odometry message ground-truth information about the state of the ego? Or is there some sort of odometry technique running under the hood that only _estimates_ the ego vehicle's state with imperfect accuracy?  Also, how is the "odom" frame that information is expressed in related to the simulator's global coordinate system?

A third question: is it possible to extract the track boundaries of the [Indianapolis Motor Speedway track](https://content.lgsvlsimulator.com/maps/indianapolismotorspeedway/) as a set of (X,Y,Z) tuples in the sim's global coordinate system?

hadiTab, 2021-02-13:
@TrentWeiss you are correct, the odometry message provides state information on the ego vehicle. We populate the `Orientation` field under `Pose` in the `Odemetery` message. It's a quaternion specifying the pose of the vehicle. It's not really specifying an "odom" frame in the traditional sense that ROS specifies. The coordinate frame specified in the message is aligned to the GPS Odometry sensor placed on the ego vehicle. The values reported are relative to the world coordinate frame.

For the third question are you referring to lane boundaries? I believe those will be available in the HD map annotations.

TrentWeiss, 2021-02-13:
> We populate the `Orientation` field under `Pose` in the `Odemetery` message. It's a quaternion specifying the pose of the vehicle. 

The quaternion piece of the pose only specifies an orientation in SO(3).  Does the sim also populate the position field of the pose with the ego's position?

> For the third question are you referring to lane boundaries? I believe those will be available in the HD map annotations.

The concept of "lane boundaries" doesn't really make sense in IndyCar racing the way it does in urban driving.  The cars in an Indy race are not limited to predefined lanes, they can go anywhere on the track at any time (the pit lane has different rules, which is the only area that can be classified as a "lane").  The only hard-limits on the actual racetrack are the inside line (inner boundary) and the outside line (outer boundary).  Would the HD map have this information?

lemketron, 2021-02-13:
As I understand it, the annotation in the IMS track is for a single lane to specify the left (inside) and right (outside) track edges, which is what allows NPCs to drive around the track. 

You should also be able to see this drivable space denoted in the segmentation camera. Try adding and enabling the visualization for the segmentation camera.

mslavescu, 2021-07-05:
@TrentWeiss I extracted the track boundaries here:

https://github.com/lgsvl/simulator/issues/1312#issuecomment-873762564