lgsvl/simulator #1252: LGSVL + API + Autoware

ajboloor, 2021-02-15:
I am trying to follow the instructions in [https://www.lgsvlsimulator.com/docs/api-how-to-run-scenario/](https://www.lgsvlsimulator.com/docs/api-how-to-run-scenario/), but I observe some errors while trying to run it.
For context, I'm trying to quite simply create a new scenario using the API to drive between two points and Autoware, doing the actual driving. Additionally, I'm trying to run everything on the same system.

Firstly, am I supposed to be running the tutorial listed above in API only mode? In that case, how do I connect it to Autoware?

When I try to do with by starting the simulator in the BorregasAve (Autoware) mode, I get a connection refused error when I try to run `python VF_S_25_Slow.py`:

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/home/adith/.local/lib/python3.5/site-packages/lgsvl/remote.py", line 28, in run
    self.loop.run_until_complete(self.process())
  File "/usr/lib/python3.5/asyncio/base_events.py", line 387, in run_until_complete
    return future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 241, in _step
    result = coro.throw(exc)
  File "/home/adith/.local/lib/python3.5/site-packages/lgsvl/remote.py", line 36, in process
    self.websocket = await websockets.connect(self.endpoint, compression=None)
  File "/home/adith/.local/lib/python3.5/site-packages/websockets/py35/client.py", line 12, in __await_impl__
    transport, protocol = await self._creating_connection
  File "/usr/lib/python3.5/asyncio/base_events.py", line 695, in create_connection
    raise exceptions[0]
  File "/usr/lib/python3.5/asyncio/base_events.py", line 682, in create_connection
    yield from self.sock_connect(sock, address)
  File "/usr/lib/python3.5/asyncio/selector_events.py", line 402, in sock_connect
    return (yield from fut)
  File "/usr/lib/python3.5/asyncio/futures.py", line 361, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 296, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/selector_events.py", line 432, in _sock_connect_cb
    raise OSError(err, 'Connect call failed %s' % (address,))
ConnectionRefusedError: [Errno 111] Connect call failed ('127.0.0.1', 8181)
```

Some advice would be appreciated.

EricBoiseLGSVL, 2021-02-16:
It looks like you cannot connect to the simulator.  Is this the correct address and port? "'127.0.0.1', 8181"

marcusvinicius178, 2021-04-07:
Hi @ajboloor  I am trying to do the same integration with PythonApi or Visual Scene Editor + AutowareAi did you solve this issue? If yes could you say me how to set these frameworks to work together...the tutorial documentation are not working for me too.

ajboloor, 2021-04-07:
@marcusvinicius178 unfortunately I was not able to get both the PythonAPI and Autoware to work at the same time with LGSVL.