lgsvl/simulator #1254: Problems with two-lane road export to OpenDRIVE map format

PavelLysenok, 2021-02-16:
Hi!

I was making some tests with exporting an OpenDRIVE map to better understand the nuances of the simulator, here are the screenshot of the test lanes and its' boundaries:

[image]
[image]

After the export I opened a .xodr file and mentioned, that these two two-laned roads were described as four one-laned roads. I checked all the boundary-lane connections and lane directions twice, but still do not understand how to fix this. 

The exported file is attached below:
[PavelLysenok_test.xodr (zip)](https://github.com/lgsvl/simulator/files/5989647/PavelLysenok_test.zip)

EricBoiseLGSVL, 2021-02-16:
I see right away that TrafficLane holder is not correct.  Traffic lanes hold the annotation objects for non intersection objects.  It should look like this.  Notice the MapLaneSection component and how all lane objects are children objects.
[image]

Please look at CubeTown to understand how basic annotation is organized

PavelLysenok, 2021-02-18:
@EricBoiseLGSVL 

Thanks! That worked.