lgsvl/simulator #1255: Running LGSVL on Windows 10 with VirtualBox from a MacBook.

ijb5117, 2021-02-18:
I'm not being able to run Simulator from the downloaded installation folder. I'm trying to open it from Windows 10 in VirtualBox.
Is there any solution to this?

lemketron, 2021-02-18:
LGSVL Simulator is only supported on Windows or Linux, and requires a fairly powerful Nvidia GPU at least 4GB (preferably 8GB) of GPU memory. For several reasons, it likely won't be possible to run this on a MacBook even in VirtualBox. Please consider different hardware such as a gaming/graphics workstation desktop or laptop (e.g. Razer Blade or equivalent with NVIDIA 10x0, 20x0, or 30x0 GPU) running your choice of Windows or Linux.

ijb5117, 2021-02-18:
Thank you for your response.

closed 2021-02-18 by lemketron