lgsvl/simulator #1256: lgsvl simulator 20.06

junhyeokgang, 2021-02-18:
hi,

i want to create new ego vehicle.
 
lgsvl simulator 20.05(maybe?) with unity 19.3.3 didn't have BaseLink.cs file

so i tried to build lgsvl simulator 20.06 release with unity 19.3.15 version  (because on unity hub, i tried to add simulator 20.06 with unity 19.3.3 but it didn't work and they say you need unity 19.3.15)

however i added simulator 20.06 with unity 19.3.15 i did't find simulator tap on menu.

how to upgrade simulator release 20.06

could you help me?

EricBoiseLGSVL, 2021-02-18:
@junhyeokgang please use the 06 release branch and Unity 2019.3.15f1
Be sure you are cloning and have git lfs installed.  Do not download the zip from github
So switch to 06 release branch, git lfs pull, pull changes and open in 2019.3.15f1
Please post any error logs or issues.  Screen shots help immensely.  Not having the Simulator tab means you have compile errors or you have issues with git lfs

junhyeokgang, 2021-02-19:
thanks a lot.