lgsvl/simulator #1257: Loosing boundary annotation exporting/importing to/from OpenDRIVE map format

PavelLysenok, 2021-02-18:
Hi again!

I faced the problem of loosing the boundary line annotation (the type of the line) after exporting and then importing a map to and then from OpenDRIVE map format. All the two boundaries of each single lane have the same (and incorrect) "LineType" values.

The map i export:

[image]

And the result of re-importing the exported map:

[image]

Is it a problem of exporting from LGSVL simulator or I have to check anything in the lines annotations?

EricBoiseLGSVL, 2021-02-18:
@PavelLysenok On 06 release we expect some manual edits to the imported map to work with the simulator NPCs and Pedestrians.  In the next release we have fixes for some issues.  Even with fixes, these images show big issues that I think may be from the hdmap itself.
1) The scale of the the environment is off.
2) The road is missing lanes and seems like it is not even the same section of road.
3) The area shown looks like a parking lot of some sort?

If it is possible, please post the hdmap so we can look at the data and we can test the latest changes to see if we can see the issue.
Adding @cyberphysicalrobotics to help as well

PavelLysenok, 2021-02-19:
@EricBoiseLGSVL 

1. Could you explain this? I'm not sure if I understand you correctly.
2. The object hierarchy in re-imported file is correct, the intersection is in it's own maternal object and all the 3 roads (one single-lane and two two-lane roads) either. But at the same time the geometry of the objects is different.
3. It's an area for self-driving cars testing, but the lanes I draw do not represent the real world - I left the pointcloud on the scene just to be able to draw new waypoints.

I also made some tests with exporting and re-importing the same map and here's what I saw:

1. Original map

[image]

2. The map I exported and re-imported from OpenDRIVE format:

[PavelLysenok_intersectiontest_OPENDRIVE.zip](https://github.com/lgsvl/simulator/files/6009495/PavelLysenok_intersectiontest_OPENDRIVE.zip)

[image]

While re-importing the map I got these notifications:

[image]

The problems are: geometry of the lanes and its' boundaries (maybe I should change the Threshold values, but, for example, when I import two CubeTown maps - from Lanelet2 and from OpenDRIVE - they are different. And the Lanelet2 map looks "much better". The screenshot is below.), the boundary type values are incorrect, the stop-line is drawn before the lane bifurcation though I haven't placed it on the map. 

[image]

3. The map I exported and re-imported from Lanelet2 format:

[PavelLysenok_intersectiontest_LANELET2.zip](https://github.com/lgsvl/simulator/files/6009547/PavelLysenok_intersectiontest_LANELET2.zip)

[image]

While re-importing the map I got these notifications:

[image]

It looks much better, but the problem with LineType values loosing is still there. I also can not understand why the map is shifted for several meters to the south. And some lanes are too simplified comparing to the original map:

[image]

4. When I tried to export the map to Apollo 5 format... I got this:

[image]

I didn't try other formats because I don't need them.

EricBoiseLGSVL, 2021-02-19:
Thanks for the images and files, this helps.
1) Ah, I see.  I thought your issue was the environment was not aligning with the annotation.

2) Good that the hierarchy is the same.  I think I see the issue.  The annotation that you import is not an opendrive/lanelet2/apollo map after import.  It gets converted into simulator annotation objects.  The importer is still in development and it is expected to need slight adjustments after import.  Intersection colliders need adjusted, stoplines need to be intersecting with the lane before the intersection, missing traffic signal data needs to be adjusted or added, very short lane sections are outright removed because simulator NPC's use the simulator annotation objects for pathing and can't navigate short lanes well.  We expect users to use their non imported hd map with their AD Stack software while importing the hdmap and making the adjustments in editor.  If you have a fake map then you need to manually create the map and then export to the AD stack hdmap.  This is not meant to be an hdmap tool for converting hdmap types.  Why are you exporting and re-importing maps repeatedly?  Each time you will need to adjust the map.  For the line type issue, we have a fix in the next release.  Also, we have requirements for the way the annotation map is created and references are assigned to correctly export to an hdmap, e.g., line types, left/right lane references in the lane component in the inspector panel/
3) see 1).  For future work, I recommend creating a simple plane in unity or modeling software, add a collider, the point cloud is confusing, esp if you are making a flat "fake" map.

4) The log warnings and logs are there to help fix issues with an import/export.  You can see that the hdmap is missing maporigin data for northing/easting.  You can see that lanes are missing data on associated lanes and ids.  You also can see it warns you about driving lanes, intersections and stoplines and they may need adjusting.  We also have documentation to help [here](https://www.lgsvlsimulator.com/docs/map-annotation/).

PavelLysenok, 2021-02-25:
@EricBoiseLGSVL Thank your answers!

2. It seems to me that I could not correctly formulate my idea. Those exports and re-imports of a very simple test map were made just for issue detection that I might face with annotating a big real-world map. As I've said, I need the map I drew in LGSVL simulator to be exported to three formats: Apollo 5, Lanelet2, OpenDRIVE. And yeah, some of the problems are connected only with importing.

3. It's an edited pointlcoud with only ground data so it works the same as just a plane.

4. Yeah, I've read it, but as I understood, it's not written for the current release version, so the issue with "ArgumentNullException" is still actual. I guess there is a problem with self reversed lanes and their "forward" and "reverse" parts connection to each other, but I checked the argument values several times and found no errors. Is it needed to describe the problem in more detail in another topic or there are any known solutions for this?

EricBoiseLGSVL, 2021-02-25:
No problem, glad to help.  Hopefully the issues are resolved in the next release.  I tested and had no errors exporting to apollo05 from the open drive file you posted (thanks for that).
[base_map.zip](https://github.com/lgsvl/simulator/files/6045826/base_map.zip)

[image]
Some changes I made before exporting to Apollo5
1) SingleLaneRoads object needs MapLaneSection.  When the importer make this object it can't parse and make a group that needs organized.
2) I extended the points for the stopline.  OpenDrive importer always makes it too small.
3) I moved all of the intersection objects out of being children of the MapIntersection_6 so I could move it over.  Then re-parented.  Importer usually doesn't align this well and always needs adjusting.  The intersection collider should always be inside the intersection and not near the stopline.  This trigger adds vehicles to the queue and will cause issues if npcs enter the queue before stopping at the stopline.  Look at CubeTown and how it works for light and stop sign intersections. 
4) If these edits still have errors, you may need to wait for the update.  It is releasing very soon :)

PavelLysenok, 2021-02-26:
@EricBoiseLGSVL 

1. Is it possible to place the MapeLaneSection objects with just one single lane in TrafficLanes parent object? Just like other double lane roads and self reverse lanes:

[image]

Or they must be "childed" exactly to SingleLaneRoads parent object? It's much easier to manage a big map when all the non-intersection lanes are in the same place.

And if it is possible, does it work with all format export (OpenDRIVE, Lanelet2, Apollo 5)?

2. Do I need to do the stoplines on every single intersection, even if I don't need them/if there are no stoplines in the real world intersection, or it's optional?

And again, does it work with all format export? Maybe it's a silly question, but according even to documentation the .unity scene in LGSVL simulator must be little bit different for different format export.

   3. I don't understand why, but this somehow worked! Thank you a lot! Moreover, this solved the same issue I had with a new bigger real world map that I've drawn "from scratch". Are there any ways to trace this issue and to understand which object should be "re-childed"?

   3. About the collider. So, if I understand you correctly, the intersection should look like this?
   
[image]

PavelLysenok, 2021-02-26:
>  I don't understand why, but this somehow worked! Thank you a lot! Moreover, this solved the same issue I had with a new bigger real world map that I've drawn "from scratch". Are there any ways to trace this issue and to understand which object should be "re-childed"?

UPD: And again, don't know why, but this problem appeared again. It appears only with two intersections with self-reversed lanes. And only sometimes - the export works with other intersections with self-reversed lanes.

[image]  [image]

I checked all the links to other self reverse lanes, all the boundary links, all the parent object - everything is like in documentation and should be okay. All the self reverse lane sections attributes look like this:

[image]
 
I also tried to delete all the self reverse annotation (to make this self-reversed lanes just simple one-way lanes) and this worked, but in documentation it's said that the self-reversed lane type is made specially for Apollo 5, so I'm not sure if such a composition will work correctly.

EricBoiseLGSVL, 2021-02-26:
@PavelLysenok Are you creating self reverse lanes for parking lots?

PavelLysenok, 2021-02-26:
@EricBoiseLGSVL No, these are just intersections for simple roads with self-reversed lanes

EricBoiseLGSVL, 2021-02-27:
We are in the midst of the new release and @cyberphysicalrobotics can't help on this yet but he will when he can.  I think we may just need to wait for release to fix this issue