lgsvl/simulator #1258: new ego vehicle(simulator release 20.06)

junhyeokgang, 2021-02-19:
unity v 19.3.15
simulator release 20.06

hi,
i tried to create new ego vehicle Following tutorial video (https://www.youtube.com/watch?v=R_Z65kimCII)

after building my new vehicle

i got this error code : 

[image]

i did googling and someone said you get a linux module.

but i remember when i installed unity 3.15, i checked both window(mono) and linux(mono).

could you please give some instruction  : (

thank you.

lemketron, 2021-02-19:
In the [building an asset](https://www.lgsvlsimulator.com/docs/assets/#building-an-asset) documentation it says:

> IMPORTANT Windows and Linux support must be installed with Unity to build asset bundles.

In the simulator [build instructions](https://www.lgsvlsimulator.com/docs/build-instructions/) documentation (just after step 20) it says:
> NOTE You will get an error when building AssetBundles if either Windows or Linux support is not installed in Unity
> Open Unity Hub and go to the Installs tab
> Click the vertical 3-dots next to the Unity version of the Simulator
> Click Add Modules
> Check Windows Build Support (Mono) or Linux Build Support (Mono)
> Click Done

junhyeokgang, 2021-02-19:
Open Unity Hub and go to the Installs tab 
-> i can't this method because when i installed unity 19.3.15 without unity hub.(my computer network has google download error) 

[image]

unity hub - >installs -> vertical 3-dots->there's no add module botton..

well, what your saying is that i reinstall unity 3.15v?  right?

thank you a lot

junhyeokgang, 2021-02-19:
i do reinstall using unity hub. that error fixed

closed 2021-02-19 by junhyeokgang