lgsvl/simulator #1259: Unable to run Lgsvl simulator " Invalid: Out of date AssetBundle, rebuild or download latest AssetBundle. "

Manoj-EK, 2021-02-19:
Hi Lgsvl team,

I have successfully created "CubeTown" map and "Jaguar XE2015 " vehicle from source as mentioned in the lgsvl documentation ( https://www.lgsvlsimulator.com/docs/build-instructions/ ) and was able to run to  LGSVL simulator successfully.

But when i added "Lincoln MKZ 2017" from lgsvl content (  https://content.lgsvlsimulator.com/vehicles/  ) and "Cubetown" map, it doesnt run simulator and throws the following error--- Invalid: Out of date AssetBundle, rebuild or download latest AssetBundle.

Could you please help me in solving this error...

Thanks in Advance

[image]

Manoj-EK, 2021-02-19:
Hi Lgsvl team,

I tried with all other vehciles from lgsvl content ( https://content.lgsvlsimulator.com/ )...It throws the same error
Only the "Jaguar XE 2015" vehicle built from source works.

Please do help in solving this issue.

Regards,
Manoj

lemketron, 2021-02-19:
What version of LGSVL Simulator are you using? Try downloading the latest release from [lgsvlsimulator.com](lgsvlsimulator.com).

Manoj-EK, 2021-02-19:
HI @lemketron ,Thank you for the quick response.

I have cloned latest version 2020.06 from lgsvl github (" https://github.com/lgsvl/simulator/tree/release-2020.06 ")

My scope of work, requires me to build it from source and then run the simulator (Downloading from official website, simulator works fine)

Please do the needful

Manoj-EK, 2021-02-22:
> 
> 
> Hi Lgsvl team,
> 
> I tried with all other vehciles from lgsvl content ( https://content.lgsvlsimulator.com/ )...It throws the same error
> Only the "Jaguar XE 2015" vehicle built from source works.
> 
> Please do help in solving this issue.
> 
> Regards,
> Manoj

Hi lgsvl team,

I have tried all possible ways like changing  vehicles and maps from lgsvl content website...Still it does not help.

Could you please help in bringing it to an logical conclusion.

I appreciate your support in Advance.

Regards,
Manoj

EricBoiseLGSVL, 2021-02-24:
@Manoj-EK are you changing simulator source code?  If so, you need to build the new assets with your new simulator.  Did you update the simulator version as well?

Manoj-EK, 2021-03-07:
@EricBoiseLGSVL   Yes, I am adding an additional sensor to lgsvl source code with appropriate steps as suggested by @Yimags
https://github.com/lgsvl/simulator/issues/1235#issuecomment-773890465 

Could you please elaborate the **building new assets with new simulator** part ?

> @Manoj-EK are you changing simulator source code? If so, you need to build the new assets with your new simulator. Did you update the simulator version as well?

yes, I have cloned latest version of lgsvl simulator 2020.06

**Thanks in Advance for your support**

EricBoiseLGSVL, 2021-03-08:
@Manoj-EK Open Simulator -> Build... at the top of the Unity Editor.  Then toggle the new sensor and click the build button at the bottom.  If you have scripts in simulator source, not in Assets/External/Sensors/YourSensor, you will need to build a new binary as well.  Glad to help and let us know if you have any issues.