lgsvl/simulator #1261: About Segmentation Camera Data

qiongwu1003, 2021-02-23:
Hi LGSVL team,

I want to confirm that the Segmentation Camera data we get in the simulator is based on the "tag" information, which we add in the map, but not calculate from the real segmentation algorithm? Am I right?
Or if I am wrong, could you please tell me the code place, thanks very much.

EricBoiseLGSVL, 2021-02-24:
Yes, the segmentation types are based on the Tag in the Unity editor.  The object that is tagged and child objects are collected and edits are made to the material to work with our edits to HDRP source.  That is why we include HDRP package in source with our edits.  This also effects Depth camera and LiDAR sensor.  The code is in EnvironmentalEffectsManager and in the HDRP source look for // LGSVL and shader.  Be sure to not have a parent object tagged and child objects tagged, Unity will throw and exception.  Be sure each tag is only for parent and child objects for each type.  You can see in CubeTown how this is setup.

qiongwu1003, 2021-02-26:
@EricBoiseLGSVL Thank you very much. The answer is very helpfull.

closed 2021-02-26 by qiongwu1003