lgsvl/simulator #1262: After importing the HD map in Open drive format, how should I use it?

hanyh1191, 2021-02-24:
I want to use the AD stack I wrote, and now I want to use the ROS2 bridge to obtain the road information, lane information, and road section information published from the simulator, so that I can query the location of each vehicle and plan the closest to the destination of my vehicle the way.
In order to obtain the above information, how can I use the high-precision map after import the HD map in Open drive format? Or are there any examples of using HD maps in lgsvl simulator?

EricBoiseLGSVL, 2021-02-25:
I'm not exactly sure what you are asking but to access the objects used to generate the HDMap and for NPC's to navigate, the MapManager.cs has all the data for the map in the scene.  This is in our Unity object/class format.  Is this what you are asking?

hanyh1191, 2021-03-01:
> I'm not exactly sure what you are asking but to access the objects used to generate the HDMap and for NPC's to navigate, the MapManager.cs has all the data for the map in the scene. This is in our Unity object/class format. Is this what you are asking?

Sorry , maybe I just want to get answers to the following questions:
How can I get the road ID, lane ID, lane segment of the traffic participant?  What's the next step after import HD map? Do I need to edit the C# source code?

EricBoiseLGSVL, 2021-03-03:
Sorry, we do not keep all the data from the hdmap in the annotation data in scene but we do have some.  We have id and connected lanes and you can see all of the data in MapLane.cs.  You could add to MapDataPoints class and on import parse the exact data you need from the hdmap, then create a sensor to parse the map data. There are many steps from just importing an HDMap to parsing unique data that you need during a simulation.  I recommend looking at docs for creating a custom sensor.

lemketron, 2021-03-03:
> I want to use the AD stack I wrote, and now I want to use the ROS2 bridge to obtain the road information, lane information, and road section information published from the simulator, so that I can [...]

The parsed OpenDrive information is not available from the simulator through the ROS2 bridge. As @EricBoiseLGSVL mentioned, you could potentially create a custom sensor to provide the road ID, lane ID, and lane segment information for each traffic participant (similar to the ground truth information but with these IDs instead of bounding boxes), but that would require additional work since that information may not actually be available (or retained) within the scene.

However, usually an AD stack imports and parses its own copy of the HD map which could then be used to determine this information for each participant (typically discovered by your perception component).

Without knowing more about what you're trying to do, it's difficult to say if you should do this in your AD stack or in the simulator, and as mentioned above, the scene may not have all the information you are looking for anyway.

hanyh1191, 2021-03-05:
@lemketron @EricBoiseLGSVL 
Thanks, I got it maybe.
What I want to do is test algorithms for behavioral decisions and path planning, rather than visually-driven autonomous driving. So, for example, if I want to use the **A-star algorithm** to find the fastest lane connection to the destination, I need follow your mentioned to do  some customizations?
In addition, The best way to use the HD map is not parse the high-precision map through the simulator and then publish the message with the sensor plugin.  Is just like you said, let AD stack import and parses HD map, to get information for each participant?

EricBoiseLGSVL, 2021-03-05:
@hanyh1191 It depends on what approach you need.  We recommend to look into the AD stack to parse this data.  You may also need to make changes in simulator but you may not.  Let us know what approach you use and any issues you have with simulator.

hanyh1191, 2021-03-06:
@EricBoiseLGSVL  I'll think about it, Thanks.