lgsvl/simulator #1263: a question about the feedback of the throttle and brake pedals

AlexandrZabolotny, 2021-02-24:
Hi team!

Help me understand!  
I am trying to run Apollo using a simulator. But I see the following trend.  

Apollo forms a "/apollo/control" channel with the following commands  
- throttle  
- brake  
- steering  

  **BUT as a result, throttle and brake often do not match the programmed value. This can be seen in the "/apollo/chassis" channel that the simulator generates.**  

For example  
For control comand Brake = 37.26...  
I get feedback - brake = 31.31...  

[image]  
[image]

hadiTab, 2021-02-26:
@AlexandrZabolotny Are you sure this isn't just due to the time difference between the two messages?

lemketron, 2021-02-26:
That appears to be a chassis message from 17sec earlier (3112-3095).

That would not be the feedback from that control message. Do you have others.

AlexandrZabolotny, 2021-03-01:
> That appears to be a chassis message from 17sec earlier (3112-3095).
> 
> That would not be the feedback from that control message. Do you have others?

[image]

[image]