lgsvl/simulator #1266: Is Apollo's auto parking feature supported with LGSVL?

ffzero1, 2021-02-25:
Hello,

I'm currently running latest LGSVL (2020.06) with Apollo 5.0. I tried AutonomousStuff Parking lot map and try to generate a route to an open parking space, but it doesn't seem to work. The vehicle only drives on the road and stop - it won't move into parking space. Is this parking feature supported in LGSVL/Apollo5.0 fork? If it does, is there anything special that I need to do? Currently, I follow steps from here https://github.com/lgsvl/apollo-5.0
Thank you in advance.

EricBoiseLGSVL, 2021-02-25:
I think it is supported but I need to check with @cyberphysicalrobotics .  Any ideas what the issue is?

lemketron, 2021-02-26:
The AutonomouStuff parking lot map was created for the Autoware Autonomous Valet Parking project which you can read about here: 
https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/avpdemo.html
https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/lgsvl.html
https://www.autoware.org/post/autoware-does-autonomous-valet-parking

Apollo also supports valet parking (https://apollo.auto/valet_parking/index.html) but it looks like the planner didn't support that until Apollo 5.5 as noted in the [Apollo 5.5 planning readme](https://github.com/ApolloAuto/apollo/blob/a29a563e95944b603ab9be338cce46e6486a89be/modules/planning/README.md). So you'll probably want to check out apollo 6.0 or apollo master which both include integrated support for LGSVL Simulator.

ffzero1, 2021-03-03:
Thank you for the information. I did upgrade to Apollo 6.0 master (following this instruction https://www.lgsvlsimulator.com/docs/apollo-master-instructions/), copied over the Autonomous Stuff parking lot map and tried. It still doesn't seem to work. It looks like I can see the parking space id but I can't create routing into the parking space. Is there any other configuration that I need to do to get this working? or should this simply work when I set the routing to open parking spot? Has anyone successfully run auto parking scenario with Apollo? Any idea/suggestion what I might be doing wrong? Thank you for the help.

[image]

[image]

lemketron, 2021-03-03:
> Has anyone successfully run auto parking scenario with Apollo?

We have not tested the Apollo parking feature; I would suggest asking in the [Apollo Issues](https://github.com/ApolloAuto/apollo/issues).

Searching for "parking" in Apollo issues, I found https://github.com/ApolloAuto/apollo/issues/13105 which suggests that parking only works on the Apollo "Sunnyvale With Two Offices" map. You'll need to create an environment in order to try that map in LGSVL Simulator. 

It might be possible to import the Apollo HD map in the next (2021) release of LGSVL Simulator and create a mesh to create a new environment for simulation.  

But it might be easier to ask in [Apollo issues](https://github.com/ApolloAuto/apollo/issues) if there's some way to make the Apollo parking feature work on the AutomomouStuff map in LGSVL Simulator.

AlexandrZabolotny, 2021-03-03:
Hi @ffzero1  
I had a similar [question](https://github.com/ApolloAuto/apollo/issues/12165)  
You can try manually setting a parking space if you want to test the parking algorithm.  

1. Find this [file](https://github.com/ApolloAuto/apollo/blob/r6.0.0/modules/routing/routing.cc)   
2. In function `bool Routing::FillParkingID(RoutingResponse* routing_response)`  add next text  
```
      const PointENU parking_point =
          routing_request.parking_info().parking_point();
      std::string parking_space_id = "PS_2";  // set parking space ID
      routing_response->mutable_routing_request()->mutable_parking_info()->set_parking_space_id(parking_space_id);
      return true;
```  
You will see that the car will build a parking trajectory and follow this trajectory.
But the trajectory repetition error will be very large. I think (but maybe I'm wrong) this is because the [settings](https://github.com/ApolloAuto/apollo/blob/r6.0.0/modules/planning/conf/planner_open_space_config.pb.txt) are not ideal for this Lincoln car  

But it is interesting to understand what needs to be done for the car to automatically look for a parking space.

ffzero1, 2021-03-03:
Thank you @AlexandrZabolotny and @lemketron.