lgsvl/simulator #1270: Map rotation and georeferencing loosing after export

PavelLysenok, 2021-03-01:
Hi team!

Thank you again for all you help, but I have another issue. 

After I export the map from Unity to Apollo/Lanelet, it rotates 90 degrees clockwise. I enter the "-90" value into Y rotation in Transform component window for all the objects (MapOrigin, MapHolder, SpawnInfo and PointCloud), but this step also changes the coordinates, I guess, and I lose the georeferencing (the exported map is shifted for 2-3 meters).

I uploaded the pointcloud and entered all the values as it's shown in YouTube tutorial video - I'm pretty sure that origin easting and northing, UTM zone are correct, and the XYZ in Transform window is 0 0 0 for the MapOrigin object.

EricBoiseLGSVL, 2021-03-01:
Is your map rotated correctly in Unity?  Please see:
Note: To ensure proper heading values in GNSS related topics, the map must be oriented in such a way the the Z-axis of the unity world coordinate is pointing East. Otherwise all orientation values will have an offset.
[here](https://www.lgsvlsimulator.com/docs/simulator-coordinate-system/).

New docs that are going to release soon that have more details but I'll add here ahead of schedule.

To ensure proper heading values in GNSS related topics, the map must be oriented in such a way the the Z-axis of the unity world coordinate is pointing East. Otherwise all orientation values will have an offset. This can be a bit unintuitive, since by default Unity will have the Z-axis pointing upward on the screen which can lead to users assuming that it aligns to the geographic North.

The images below illustrate the map of GoMentum Station with the correct orientation in Unity compared to a satellie image of the area from Google Maps:
[image]
[image]

As evident above, upon the first glance it appears that the GoMentum Station map in Unity is oriented incorrectly; however, if the user were to rotate the view-point so that the Z-axis would be pointing to the right-side of the screen (where the geographic East normally points), the two maps would align as seen in the image below:

[image]

funix0, 2021-05-07:
hi，When I import OpenDrive maps, I export Apollo maps. I had the same problem. How did you solve it?