lgsvl/simulator #1271: Segmentation Camera image with and without Instance Segmentation

FabianG92, 2021-03-01:
Hi!

I'm trying to get Images from the Segmentation Camera and I'm having a bit of trouble.
My problem is, I'm trying to get a image of just semantic segmentation from one Sensor and another image of semantic segmentation and instance segmentation from another. 
I tried to add the sensor twice to the vehicle configuration (it's posted below), with just one sensor having the InstanceSegmentationTags. I changed the Min Distance for the second sensor to see if it would affect both sensors aswell (which doesn't seem to happen) and to see if it would create atleast different image (which it does). 
Im getting the images through the Python API and while loading the sensors both show up individually aswell.

`[
  {
    "type": "IMU",
    "name": "IMU",
    "params": {
      "Topic": "/imu",
      "Frame": "imu"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "Lidar",
    "name": "Lidar",
    "params": {
      "LaserCount": 32,
      "MinDistance": 0.5,
      "MaxDistance": 100,
      "RotationFrequency": 10,
      "MeasurementsPerRotation": 360,
      "FieldOfView": 41.33,
      "CenterAngle": 10,
      "Compensated": true,
      "PointColor": "#ff000000",
      "Topic": "/point_cloud",
      "Frame": "velodyne"
    },
    "transform": {
      "x": 0,
      "y": 2.312,
      "z": -0.3679201,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "Segmentation Camera",
    "name": "Segmentation Camera",
    "params": {
      "Width": 1920,
      "Height": 1080,
      "Frequency": 15,
      "FieldOfView": 50,
      "MinDistance": 0.1,
      "MaxDistance": 2000,      
      "Topic": "/simulator/segmentation_camera"
    },
    "transform": {
      "x": 0,
      "y": 1.7,
      "z": -0.2,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
{
    "type": "Segmentation Camera",
    "name": "Instance Segmentation",
    "params": {
      "Width": 1920,
      "Height": 1080,
      "Frequency": 15,
      "FieldOfView": 50,
      "MinDistance": 10,
      "MaxDistance": 2000,
      "InstanceSegmentationTags": [
        "Car",
        "Pedestrian"
      ],
      "Topic": "/simulator/segmentation_camera"
    },
    "transform": {
      "x": 0,
      "y": 1.7,
      "z": -0.2,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
},
  {
    "type": "Color Camera",
    "name": "Main Camera",
    "params": {
      "Width": 1920,
      "Height": 1080,
      "Frequency": 15,
      "JpegQuality": 75,
      "FieldOfView": 50,
      "MinDistance": 0.1,
      "MaxDistance": 2000,
      "Topic": "/simulator/main_camera",
      "Frame": "camera",
      "Distorted": true,
      "DistortionParameters": [
        -0.25349,
        0.11868,
        0,
        0
      ]
    },
    "transform": {
      "x": 0,
      "y": 1.7,
      "z": -0.2,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  }
]`

This one has the instance segmentation and a Min Distance set to 10.
[image]
This one has the instance segmentation aswell but shouldn't have it.
[image]

Is it possible to get the images with one actually having just semantic segmentation and the other one having both? 
Or am I just having a complete wrong thought about how to do it.

Thank you for your answer

lemketron, 2021-03-01:
It looks like you're having both sensors publish to "Topic":"/simulator/segmentation_camera"

Can you try setting the instance segmentation camera instance to publish on a different topic? If that doesn't work for you then what happens if you only use one or the other but not both at the same time?

EricBoiseLGSVL, 2021-03-01:
@FabianG92 Sorry we don't support instance segmentation for just one camera and non instance segmentation on others.  Once you enable instance segmentation, it applies to all materials in the simulation.  It really isn't possible with the way segmentation works.

FabianG92, 2021-03-01:
> @FabianG92 Sorry we don't support instance segmentation for just one camera and non instance segmentation on others. Once you enable instance segmentation, it applies to all materials in the simulation. It really isn't possible with the way segmentation works.

Alright thank you for your help.

closed 2021-03-01 by FabianG92