lgsvl/simulator #1272: Asset bundle out of Date

jeniwang, 2021-03-02:
So we built a track on unity and we need to integrate it onto the simulator. However we noticed that when we followed the instructions to build simulator from source: https://www.lgsvlsimulator.com/docs/build-instructions/
When we ran the simulator from these instructions, we got an error saying that the asset bundle was out of date even on the sample ones from the content website of the official website. This problem didnt occur when we just directly downloaded the simulator and unzip and ran the executable but since we are using a track built on unity we need to be able to build the simulator from unity which is currently causing the out of date issue.

EricBoiseLGSVL, 2021-03-03:
Did you clone the 06 release branch of simulator?  Do not use master for latest releases.

jeniwang, 2021-03-03:
How do i make sure i clone the 06 branch i just followed instructions on
the site and it said to clone —single branch im not sure what this means as
i didnt see a branch on the git called single branch

lemketron, 2021-03-03:
To clone 2020.06, try this:
`git clone git@github.com:lgsvl/simulator.git --branch release-2020.06 --single-branch`

EricBoiseLGSVL, 2021-03-03:
Sorry this is outdated, @lemketron is right use that command and be sure you have Unity 2019.3.15f1 Please see our releases notes [here](https://github.com/lgsvl/simulator/releases/tag/2020.06)

You are probably in 2019.3.3 which is why you get asset bundle out of date errors.  We are releasing a new update and all docs will be updated as well.