lgsvl/simulator #1273: No transform from base_link to world

Johanfanas, 2021-03-02:
Hi,

I followed the instructions from the LBSVL simulator correctly, but I am getting a warning on Rviz that it can't find the transform between base_link to world. There is also another warning for the velodyne and world but I assume that one is going to go off after the base_link to world gets fixed. Rviz cant find the urdf file for the vehicle as well.

I am running Autoware with the LGSVL simulator on Ubuntu 18.04. Also, I am running the Jaguar2015XE in BorregasAve.

Any suggestions how to fix this ?

[image]

[image]

hadiTab, 2021-03-03:
@Johanfanas which autoware are you using? Is this Autoware Ai? If so which version? Can you also send a link to the instructions you followed?

From the information provided, it seems that localization is either not working properly or has not been activated.

Johanfanas, 2021-03-05:
I am using Autoware AI version 1.14.0 (I believe that's the one on your documentation website). The instructions I followed are here https://www.lgsvlsimulator.com/docs/autoware-instructions/