lgsvl/simulator #1277: problem about clusters

duanchengwen, 2021-03-03:
how to assigning different sensors to each client for simulation (including itself)

for example:
i want to enable imu and lidar for the first client, and enable camera for the second client

lemketron, 2021-03-03:
You can configure multiple ego vehicles, and each vehicle can have its own unique sensor configuration and (ROS, Cyber, etc.) bridge where that sensor data will be sent. 

See [vehicle configuration](https://www.lgsvlsimulator.com/docs/vehicles-tab/), [simulation configuration](https://www.lgsvlsimulator.com/docs/simulations-tab/) and [sensor configuration](https://www.lgsvlsimulator.com/docs/sensor-json-options/) documentation for more details.

duanchengwen, 2021-03-03:
> You can configure multiple ego vehicles, and each vehicle can have its own unique sensor configuration and (ROS, Cyber, etc.) bridge where that sensor data will be sent.
>
> See [vehicle [...]

My requirement is to deploy multiple sensors in the same car. I deploy imu, color camera, lidar in the car, but the imu message cannot reach 100hz, And there will be a situation where the timestamps of the message headers of multiple frames are consistent, No matter if I run lgsvl simulator on 1080TI graphics card or 3070 graphics card.
[image]

[image]
Is there any way to avoid this problem

MisterioRemo, 2021-03-26:
@duanchengwen hello.

> how to assigning different sensors to each client for simulation (including itself)

I belive it's not possible right now.
Distribution into clusters occurs in _SensorsController.cs_ in `void DistributeSensors()` function. Perhaps you can modify it yourself.