lgsvl/simulator #1278: Unable to access CAN-Bus data via Python API

cubrink, 2021-03-03:
I am unable to access certain attributes of vehicles using the Python API. Specifically, I would like to read values for acceleration, steering and braking. 

I believe the [CAN-Bus sensor](https://www.lgsvlsimulator.com/docs/sensor-json-options/#can-bus) should provide this information but I can only access the [frequency](https://github.com/lgsvl/PythonAPI/blob/master/lgsvl/sensor.py#L134) of the sensor via the Python API.  This appears to be a known issue #1144. 

First I tried modifying the Python API so I could access the full response, but was still only given the frequency of the sensor.

After that didn't work I traced the data back through the code and I believe the problem is that the API command for the CAN-Bus only adds the frequency as to the [response](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Api/Commands/VehicleGetSensors.cs#L113). The `CanBusSensor` appears to have a `CanBusData` member variable which [contains the information I need](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/CanBusSensor.cs#L76), but it isn't included in the response.

Have I overlooked a way to access this data with the Python API? I didn't see any [API calls](https://github.com/lgsvl/simulator/tree/master/Assets/Scripts/Api/Commands) that would access this data, though its possible I could have missed the one I want.

lemketron, 2021-03-03:
The Python API is not intended for real-time polling of sensor data. Sensor data should be published to a bridge (ROS, ROS2, CyberRT, etc.) where it can be monitored, collected, and/or processed while the simulation is running, either in real-time or in (stepped) simulation time using the simulator clock sensor.

cubrink, 2021-03-03:
Thanks for the quick reply. The issue I had linked to mentioned that the next release would make some of the sensor data available to the Python API. Is this no longer the case?

EricBoiseLGSVL, 2021-03-03:
Yes, sorry for the confusion.  After reviewing all the sensors that would need to be added and the performance issue, we decided to not support api for data collection. Please use the bridge for data collection.  You can change the source to do this but we do not recommend.  You will need to change the python api and api commands in simulator source.