lgsvl/simulator #1286: Get Camera output stream and pipe it into an input stream of python

cryptec95, 2021-03-05:
Hello guys,

i wonder if theres a way to get the camera video stream and pipe it to a python program input videostream?

i saw that you can captur images with the python API but i need a realtime videostream.

Thanks for your help

EricBoiseLGSVL, 2021-03-05:
@cryptec95 Since Python API is not a good way to process data in real time, we created the bridge to handle this.  You can create a sensor that can process the video, we have an example that saves video in the next release but it does not send this data over bridge yet.  You could alter this sensor to send the video.

cryptec95, 2021-03-06:
@EricBoiseLGSVL where i can find a documentation about the bridge you mentioned?

EricBoiseLGSVL, 2021-03-08:
@cryptec95 [Here](https://www.lgsvlsimulator.com/docs/lgsvl-msgs/) are lgsvl bridge msg types