lgsvl/simulator #1290: Traffic lights Lanelet2 export issue

PavelLysenok, 2021-03-10:
Hi team!

I cannot export the traffic light to Lanelet2 map format. 

Tried to create a small map with just one lane to test the export - but I see no regulatory_elements in the .osm file. The pole is created and the mapsignal object is assigned it. The stopline is assigned to the lane.

[image]

Is this feature available now? And if not, would it be possible to export traffic lights to Lanelet2 format in the next release?

EricBoiseLGSVL, 2021-03-11:
Yes we support traffic signal export to Lanelet2 format and it contains regulatory elements in the .osm file.  Please see cube town lanelet2 exported map.  Do you have errors when exporting your annotations from simulator?  Is this an issue when you are importing the map back into simulator?

PavelLysenok, 2021-03-12:
@EricBoiseLGSVL Yeah, I've checked almost all the examples from the LGSVl website, but it's still not clear how to assign maplanes, stoplines, poles, traffic signal objects and their meshes to each other. In comparison with the examples I annotated the map in a right way. When I export the same map to Apollo5, the traffic lights are on the right places. So, as I guess, It's only a problem with Lanelet2 format. 

No, there are no errors in exporting. Just "Successfully generated and exported Lanelet2 map" message. 

No, it's issue only with export. The map on the screenshot was drawn and annotated just in Unity. To check the result of export to Lanelet2 I use JOSM (for geometry) and just a simple text editor (for attributes, values etc.).

EricBoiseLGSVL, 2021-03-13:
@PavelLysenok This maybe a bug.  Hopefully our next release fixes this issue with LaneLet2 export

MapSignal finds the closest SignalLight class within 0.1 meters and assigns that, on export it looks for the bounds of MapSignal.
MapSign needs the stopline assigned in the class.  Poles, Maplanes, MapLines etc have the data pulled from them to export to the chosen format (Unity annotation has multiple data fields to support all the different exporters)  You can pull up the code for each exporter, e.g. Lanelet2MapExporter.cs and see how it parses the annotation data to export to that hdmap type.  Assigning to meshes is really only for NPC behavior during a simulation, not the exporter.

closed 2021-07-16 by EricBoiseLGSVL