lgsvl/simulator #1291: Pulling Data from 3D Ground Truth Sensor

efj5n8, 2021-03-10:
I recently implemented the 3D Ground Truth Sensor (https://www.lgsvlsimulator.com/docs/modular-testing/ ) in LGSVL due to issues with the perception module in Apollo. For my research I need the ability to alter the data that the ego vehicle sees (i.e. what the ground truth sensor in LGSVL feeds the ego vehicle). Is there any way to access/manipulate the data from the 3D Ground Truth Sensor. For reference, I am using a 2 machine setup with the following specs
[image]

lexob9, 2021-03-11:
how did you start the apollo 6?

i started it with these commands:
```
./apollo.sh build_opt_gpu
./scripts/bootstrap_lgsvl.sh
```

but Localization module doesn't work in Mkz Standard debug mode. It only works in Mkz LGSVL mode.

EricBoiseLGSVL, 2021-03-11:
@efj5n8 Yes, you can modify the ground truth sensor to achieve what you want.  The sensor is just a Trigger volume that adds objects that are within to the data.  You can change the way these objects are detected so then the data sent is what you need.  Not knowing how you plan to alter the data makes it difficult to point you in the right direction.

lemketron, 2021-03-12:
> For my research I need the ability to alter the data that the ego vehicle sees (i.e. what the ground truth sensor in LGSVL feeds the ego vehicle). Is there any way to access/manipulate the data from the 3D Ground Truth Sensor.

Are you saying you want to modify the 3D ground truth sensor messages before they get to Apollo?  You can change the topic from `/apollo/perception/obstacles` to something else like `/apollo/perception/obstaclesFromSim` and create a CyberRT component that gets messages from the simulator, modifies the messages, and then publishes them on `/apollo/perception/obstacles`.

If you need access to "the world" inside of the simulator, then you'll need to modify the 3D Ground Truth sensor in the simulator (or create a new sensor based on it) to do whatever you want.

efj5n8, 2021-03-12:
@EricBoiseLGSVL  Sorry for the delayed response. I need to alter the data to simulate a cyber attack on the ego vehicle. One of the possible attack vectors would be the perceived speed/distance/other data of objects around the ego vehicle. So I would need to be able to alter the so that the ego vehicle in the sim thinks that the object in front of it is 15 m away when it is actually 5 and then go from there. @lemketron Based on the scenario above, which of the two approaches that you mentioned would be best? Thank you for all of the help!

EricBoiseLGSVL, 2021-03-13:
If you are trying to simulate an attack, it depends on the vector it originates.  I would alter the data at the sensor.  You could even open a channel of attack on the sensor itself from a UDP connection.  That way the simulated attack would be on the sensor level.  Now if you want a middle man attack on the transfer of data from the sensor to AD stack, then @lemketron idea would be the way to go.

closed 2021-04-26 by efj5n8