lgsvl/simulator #1292: Apollo modules crash because of not enough memory on RTX 2080 8gb

lexob9, 2021-03-10:
Hello,

It was stated in documentations and in forums that Apollo master when run on a separate computer from LGSVL, should be ok on 8gb video card. But soon after enabling modules, some of them crash, usually Perception or Traffic Lights. When at the same time the video memory usage was monitored with nvidia-smi, it was observed that maximum memory is reached and that causes modules to get turned off.

Could you please advise if it has any sense to troubleshoot this case with this video card?

System:
OS: Ubuntu 20
RAM: 16gb
VGA: RTX 2080 8gb

lemketron, 2021-03-10:
Apollo master has known issues with perception (and traffic light) so we suggest using [modular testing](https://www.lgsvlsimulator.com/docs/modular-testing/) as a workaround until it is resolved.  

Please see https://github.com/ApolloAuto/apollo/issues/13488

lexob9, 2021-03-11:
I have tried modular testing example with Apollo 6.0. But couldn't manage to make localization mode work in Mkz Standard debug mode. It only works in Mkz LGSVL mode.

There is a broken link in the manual that you suggested on the marked spot of the picture below, that should in theory tell a way to start apollo the right way for this example:
[image]

Right now i started apollo like this, from the main manual:
./apollo.sh build_opt_gpu
./scripts/bootstrap_lgsvl.sh

Should it be started any other way?

lexob9, 2021-03-20:
starting the apollo with ./apollo.sh build_gpu instead of ./apollo.sh build_opt_gpu helped in my case. i don't know if it is a coincidence or not.

closed 2021-03-20 by lexob9

lemketron, 2021-03-22:
> starting the apollo with ./apollo.sh build_gpu instead of ./apollo.sh build_opt_gpu helped in my case. i don't know if it is a coincidence or not.

I'm wondering if that helped more because it forced a clean re-build rather than because it is a non-optimized (debug) build, but if it's working for you then great! We usually use build_opt_gpu.