lgsvl/simulator #1293: Crash when launching self-compiled simulator\

ZiwenWan, 2021-03-11:
Hi LGSVL team,
I met some problems when I tried to simulate on a map built by my self.
Here are what I did on my Windows machine:
- Pull the simulator code from GitHub and checkout to the release-20.06 branch
- Run the command 'git lfs pull'
- run 'npm install' for WebUI
- Open the project with Unity 2019.3.15f1
- Build the WebUI
- Build both the map and simulator (I built for both Linux and Windows)
Here is my problem:
- The released binary can launch successfully on both windows and Linux, but the map generated can not be loaded successfully. I got the error message that the map can not be loaded.
- When I launched the self-built executable on Windows and Linux, the program will crash after showing the logo of Unity. I could not see the button "Open browser"
- Here is the player log when running the self-built executable on Windows:
 

> Mono path[0] = 'C:/Users/Ziwen/Simulator/SLR_bicycle/simulator_Data/Managed'
> Mono config path = 'C:/Users/Ziwen/Simulator/SLR_bicycle/MonoBleedingEdge/etc'
> Loading player data from C:/Users/Ziwen/Simulator/SLR_bicycle/simulator_Data/data.unity3d
> Initialize engine version: 2019.3.15f1 (59ff3e03856d)
> [Subsystems] Discovering subsystems at path C:/Users/Ziwen/Simulator/SLR_bicycle/simulator_Data/UnitySubsystems
> GfxDevice: creating device client; threaded=1
> Direct3D:
>     Version:  Direct3D 11.0 [level 11.1]
>     Renderer: NVIDIA GeForce RTX 2060 (ID=0x1f08)
>     Vendor:   
>     VRAM:     5980 MB
>     Driver:   27.21.14.5167
> Begin MonoManager ReloadAssembly
> - Completed reload, in  1.066 seconds
> WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
> WARNING: Shader Did you use #pragma only_renderers and omit this platform?
> D3D11 device created for Microsoft Media Foundation video decoding.
> <RI> Initializing input.
> 
> New input system (experimental) initialized
> <RI> Input initialized.
> 
> <RI> Initialized touch support.
> 
> WARNING: Shader Unsupported: 'Shader Graphs/VehicleLightsShader' - All passes removed
> WARNING: Shader Did you use #pragma only_renderers and omit this platform?
> WARNING: Shader Unsupported: 'Shader Graphs/CloudGenerator' - All passes removed
> WARNING: Shader Did you use #pragma only_renderers and omit this platform?
> UnloadTime: 0.731700 ms
> Null Sensor Prefab Detected - Check RuntimeSettings SensorPrefabs List for missing Sensor Prefab 
> (Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Setting up 3 worker threads for Enlighten.
>   Thread -> id: 1034c -> priority: 1 
>   Thread -> id: 21d8 -> priority: 1 
>   Thread -> id: d828 -> priority: 1 
> 
Some other information: 
- I checked the CPU driver is correct by nvidia-smi. Also, the released executable can work perfectly. 
- I checked that libvulkan is installed by command vulkaninfo
- I also confirmed that there is enough space on GPU
Thanks a lot!

EricBoiseLGSVL, 2021-03-11:
```
Null Sensor Prefab Detected - Check RuntimeSettings SensorPrefabs List for missing Sensor Prefab
```
This is your issue.
Find `RuntimeSettings.asset` in the editor
Open SensorPrefabs list and make sure no sensors are null or missing.  If you have created a new sensor in the simulator source code, be sure it is listed here.
Let us know if you have any other issues.
[image]

ZiwenWan, 2021-03-11:
Thanks for the answer. I followed your instruction and I found that element 21 (WheelControlSensor in your screenshot) in the runtime setting is missing. Can you provide more details about how to fix the missing sensor? 

Also, I have not added any sensors. Do you have any idea of why this happens?

Thanks!

EricBoiseLGSVL, 2021-03-12:
Unity can be a pain sometimes.  I usually see this when I pull changes or switch branches while Unity is open.  I try to always close unity when using git and open back up once I am finished.  esp if you don't force compile with Ctrl+R.  I always set unity to not recompile until I say.  To fix, just drag the wheel control sensor prefab into the missing spot.

ZiwenWan, 2021-03-12:
Thanks for the reply. I followed the instruction you gave:

- import the asset of wheelcontrolsensor
- Drag the prefab to the missing element
Now, the crash problem has been fixed and the simulator can launch perfectly. I still met some problems when I tried to run the simulation with the map built locally. 

Here is the snapshot of the error message:
[image]

And here is the log:

> Mono path[0] = 'C:/Users/Ziwen/Simulator/SLR_bicycle/simulator_Data/Managed'
> Mono config path = 'C:/Users/Ziwen/Simulator/SLR_bicycle/MonoBleedingEdge/etc'
> Loading player data from C:/Users/Ziwen/Simulator/SLR_bicycle/simulator_Data/data.unity3d
> Initialize engine version: 2019.3.15f1 (59ff3e03856d)
> [Subsystems] Discovering subsystems at path C:/Users/Ziwen/Simulator/SLR_bicycle/simulator_Data/UnitySubsystems
> GfxDevice: creating device client; threaded=1
> Direct3D:
>     Version:  Direct3D 11.0 [level 11.1]
>     Renderer: NVIDIA GeForce RTX 2060 (ID=0x1f08)
>     Vendor:   
>     VRAM:     5980 MB
>     Driver:   27.21.14.5167
> Begin MonoManager ReloadAssembly
> - Completed reload, in  1.551 seconds
> WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
> WARNING: Shader Did you use #pragma only_renderers and omit this platform?
> D3D11 device created for Microsoft Media Foundation video decoding.
> <RI> Initializing input.
> 
> New input system (experimental) initialized
> <RI> Input initialized.
> 
> <RI> Initialized touch support.
> 
> WARNING: Shader Unsupported: 'Shader Graphs/VehicleLightsShader' - All passes removed
> WARNING: Shader Did you use #pragma only_renderers and omit this platform?
> WARNING: Shader Unsupported: 'Shader Graphs/CloudGenerator' - All passes removed
> WARNING: Shader Did you use #pragma only_renderers and omit this platform?
> UnloadTime: 1.338500 ms
> Loaded 3 NPCs behaviours and 6 NPC models in 00:00:00.0266060 
> (Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Current Database Version: 3 
> (Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Final Database Version: 3 
> (Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Setting up 3 worker threads for Enlighten.
>   Thread -> id: ca48 -> priority: 1 
>   Thread -> id: 4380 -> priority: 1 
>   Thread -> id: d748 -> priority: 1 
> Getting current user 
> (Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Listing maps 
> (Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Listing simulations 
> (Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Starting simulation with id 13 
> (Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Unable to read header from archive file:  
> (Filename:  Line: 74)
> 
> Unable to read header from archive file:  
> (Filename:  Line: 74)
> 
> Failed to start 'SLR_bicycle' simulation 
> (Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Exception: Failed to load environment from 'SingleLaneRoadBicycle' asset bundle
>   at Simulator.Loader+<>c__DisplayClass36_0.<StartAsync>b__0 () [0x00716] in <479212dac4bd47bfaa5220c74822c82c>:0 
> UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
> UnityEngine.DebugLogHandler:LogException(Exception, Object)
> UnityEngine.Logger:LogException(Exception, Object)
> UnityEngine.Debug:LogException(Exception)
> Simulator.<>c__DisplayClass36_0:<StartAsync>b__0()
> Simulator.Loader:Update()
>  
> (Filename: <479212dac4bd47bfaa5220c74822c82c> Line: 0)
> 

BTW, I think I executed the git pull command when the unity was still running. I was wondering whether you can provide more tips on how to recompile the project? I think that may help. 

Thanks a lot!

EricBoiseLGSVL, 2021-03-12:
Is the map scene file named SingleLaneRoadBicycle and the repo folder named SingleLaneRoadBicycle? See CubeTown:
[image]

Please open the bundle with any zip application and screen shot the contents

ZiwenWan, 2021-03-13:
SingleLaneRoadBicycle is just the nickname used in the simulator to refer to the map, right? I am not sure why I have to make the name consistent. The map scene file name and the asset bundle name are both "SingleLaneRoad". I used the project pulled from Github and just added another asset bundle into the map. 

Here is the screenshot of the bundle:
[image]

Thanks!

EricBoiseLGSVL, 2021-03-14:
If you are creating SingleLaneRoad, then keep this name.  If you want a new one, make a new repo. You probable are having name conflicts with the map bundle being the same name.  How are you adding another asset bundle into the map?

ZiwenWan, 2021-03-15:
Here is a list of what I did to modify the map:
- Import an asset from asset store in Unity
- Directly drag the prefab into the map and adjust the position
- Rebuild the map
I did not aim to build a new map. I just made some changes and rebuilt the single-lane road map.
Thanks!

EricBoiseLGSVL, 2021-03-15:
Do you have code components on the prefab you dragged in?  Did you move the asset store assets into the folder for the environment repo?