lgsvl/simulator #1294: Exception: PLY file has unsupported property type int

Michealhao1991, 2021-03-12:
hello, I can not import ply file ,and there is a error in editor
[image]

EricBoiseLGSVL, 2021-03-13:
@kamkolak Is this fixed in latest as far as you know?
@Michealhao1991 can you post the file here for us to try parsing it?

Michealhao1991, 2021-03-14:
> @kamkolak Is this fixed in latest as far as you know?
> @Michealhao1991 can you post the file here for us to try parsing it?

I downland from this adress: https://github.com/lgsvl/simulator/issues/1294#issuecomment-797828337

Michealhao1991, 2021-03-14:
ply adress  : https://www.cc.gatech.edu/projects/large_models/dragon.html
I try to convert ply to pcd  and  it worked.

EricBoiseLGSVL, 2021-03-14:
Ah, yes we have updates in the next release to support .ply files but pcd are only working in this release

Michealhao1991, 2021-03-15:
> Ah, yes we have updates in the next release to support .ply files but pcd are only working in this release

I found another problem in my pointcloudRender ，there is no vertical mesh was generated:
[image]

EricBoiseLGSVL, 2021-03-15:
@Michealhao1991 what do you mean vertical mesh?  The point cloud is rendering the data provided.  Maybe the conversion to pcd is not working correctly?

closed 2021-07-16 by EricBoiseLGSVL