lgsvl/simulator #1295: "Invalid: Unsupported message type Simulator.Bridge.Data.SignalDataArray used for CyberRT bridge"

1power1, 2021-03-15:
Hi team! Help me please!
I want to use LG and apollo for co-simulation, but I modified the map environment, NPC vehicles and EGO vehicles myself. In order to ensure that the changes take effect, I recompiled the binary file of the simulator, so far there is no problem.
But when I tried to add 3D Ground Truth and Signal sensors to the JSON configuration using Modular testing, it would remind me of the error "Invalid: Unsupported message type Simulator.Bridge.Data.SignalDataArray used for CyberRT bridge", but when I posted it When running on the 2020.06. version, it can run normally, but when running on the 2020.06 version, the NPC vehicle will not change according to my modification.
I would like to ask, how can the modification of the map environment, NPC vehicles, EGO vehicles and Modular testing take effect at the same time? Or is this feature not currently available?

EricBoiseLGSVL, 2021-03-15:
What changes did you make?

1power1, 2021-03-16:
> What changes did you make?
I modified the model in the SanFrancisco scene in the sumilator project in unity3d (for example, deleted some old buildings, added some new buildings), and modified the mesh grid of NPC vehicles (also re-engineered NPCmanager and NPCsettings). Configuration), the model of the EGO vehicle has been modified to simulate the environment I need. Finally, the map environment and EGO vehicles are built, and the Simulator executable file is run in the corresponding folder. When the default JSON sensor configuration is used, it runs without any problems, and all modifications can run normally.
But when I added 3D Ground Truth and Signal sensors to the JSON sensor configuration to perform co-simulation with apollo6.0, the Simulator executable file did not run normally, and it prompted "Invalid: Unsupported message type Simulator.Bridge.Data.SignalDataArray used" for CyberRT bridge" error.
I would like to ask how to add 3D Ground Truth and Signal sensors to my modified project?
[image]
[image]
[image]

EricBoiseLGSVL, 2021-03-17:
Did you create the protobuf messages for the new data?

1power1, 2021-03-17:
> Did you create the protobuf messages for the new data?
I think I shouldn't do this step. In which file should I add protobuf messages with new data?

EricBoiseLGSVL, 2021-03-17:
@hadiTab Do you have the link to protobuf conversion tool?

1power1, 2021-03-23:
> @hadiTab Do you have the link to protobuf conversion tool? @1power1 You have to do this. We provide basic sensors and have created the protobuf conversions for them. If you want to add your own for CyberRT, you must create these files, like we have in simulator source code. We will try to include better documentation on this in the future. In the meantime, please look into Apollo docs for protobuf.

If it is not convenient to provide the protobuf conversion tool, can you tell me which files or data in the script should be modified? I really need to add 3D turth and Signal sensors to the modified scene, thank you again.

EricBoiseLGSVL, 2021-03-24:
We have a link to the tool, but it is very old.  We will find it, no worries.

EricBoiseLGSVL, 2021-03-24:
Here is the tool.  One of our engineers had to dig up the tool from wayback machine since the apollo link was broken.
[protogen 2.3.16.zip](https://github.com/lgsvl/simulator/files/6199127/protogen.2.3.16.zip)

1power1, 2021-03-25:
After I decompressed the files in the link, there are two folders. The files in these two folders do not seem to be resource files in the simulator/Assets. When I try to put these two folders into the simulator/Assets, the entire The project crashed, and there seems to be no such files in the folder where the executable files have been built. Could you tell me the specific usage and steps of these files? Is there anything common to windows and linux platforms? thank you very much!
[image]

EricBoiseLGSVL, 2021-03-25:
No, this is a tool to convert msg to protobuf format.  In fact it's not even an apollo tool, sorry for the confusion.  Please look up documentation on this [here](https://www.nuget.org/packages/protobuf-net.Protogen/2.3.16).  This package was the only one that supported the Unity .net version that we were using at the time.  We will be updating this as soon as we can.

Czl0223, 2021-03-29:
> No, this is a tool to convert msg to protobuf format. In fact it's not even an apollo tool, sorry for the confusion. Please look up documentation on this [...]

I also encountered the same problem. I would like to ask about the files that need to be converted, including those, and how to use the protogen.2.3.16 conversion tool, and where should the converted files be placed? I haven't used Unity3D for message format conversion, so these problems may seem strange, but I hope you can describe them in as much detail as possible. thank you very much.

EricBoiseLGSVL, 2021-03-29:
You can see the ROS2 conversions that we created in source.  CyberRT requires these protobuf file conversions to work with bridge.  You can see the protobuf converted files in simulator source as well.  The github link above, for the tool, walks through the requirements and process to convert the data types.  We admit this is not the best documentation and we are adding improvement and documentation to our roadmap for this feature.